iNavFlight/inav #137: Reduce number of variables written into Blackbox

DzikuVx, 2016-04-01:
With new data added to Blackbox in iNav, BB performance went down considerably. With the same setup in CF I'm able to record 1:1 at 2500 looptime. In iNav, with looptime 2000 it is possible to record at 1:3 ratio despite increased port speed. Same cards, same OpenLog writer.

Personally I do not think all recorder sensors are important. For example:
* raw accelerometer
* raw magnetometer
* heading[roll]
* heading[pitch]

iNav specific entries are different story.

What do you think?

digitalentity, 2016-04-01:
@DzikuVx 
Indeed, there are too many blackbox variables. However I think raw sensor data may be useful for debugging issues with sensor noise etc. Heading[] is not logged it's calculated by blackbox viewer.
We should carefullt think about what variables to remove without compromising the usefulness.

DzikuVx, 2016-04-01:
Yes, Ticket is mostly to remeber

theArchLadder, 2016-04-01:
How about an option to do "lightweight blackbox logging" so that the logging can be done 1:1?

When doing pid tuning, checking for vibration and stuff like that, 1:1 logging is desired. Logging gyro, acc, pid-stuff and motors would almost be enough here i think, and should allow 1:1 logging.

When logging for navigation stuff, a slower logging rate is acceptable IMHO, so just log everything, but at a slower rate, i use 100hz right now. I can't think of anything that is not useful right now, i find the raw sensor data etc useful, i like to have it all :)

@digitalentity what do you think is the slowest logging rate that will still be useful for inav related stuff? Would even as low as 10hz still be useful?

I know that we would like to both eat the cake and have it. But i think this could be a good compromise if that's not possible.

stronnag, 2016-04-01:
With a reasonable SDHC card, one can do 1:1 at practical inav rates (2000 loop time) already.
````
Data rate  497Hz  23364 bytes/s     233700 baud
11 frames failed to decode, rendering 106 loop iterations unreadable. 121 iterations are missing in total (242ms, 0.26%)
````
even on F1, sub €7.5 16GB card (Transcend Ultimate 16GB Micro SD (SDHC) Card - Class 10)

closed 2017-04-04 by digitalentity