iNavFlight/inav #138: FW - Possibility to have airplane always ARM'ed

digitalentity, 2016-04-01:
Some safety measures must be considered:
 * MOTOR_STOP must be selected
 * Home position should be set when GPS is firstly acquired

oleost, 2016-04-01:
How to handle failsafe?

Example for me: when the TX and RX are close I loose the RC link( bending over to pick-up plane ). Because it's disarmed it's doesn't enter failsafe in this case.

How to minimize risk of this happening with always armed? Since you often land some distance from home.

How to other FC do this?

oleost, 2016-04-01:
*doesn't enter failsafe stage 2

melkati, 2016-04-01:
Others FC have into account speed and height (if speed < x kmh or eight bellow x meters doesn't engage RTH).

oleost, 2016-07-20:
#310 

Maybe close this and let #310 take it forward?

closed 2016-07-21 by digitalentity