iNavFlight/inav #139: Reduce number of available mixers

DzikuVx, 2016-04-01:
There are many exotic mixers like BiCopter, Dualcopter or SingleCopter that are not popular. Keeping that configuration is wast of resources. And they can be entered manually with custom mixer.
It is impossible to prevent users form selecting then since list is hardcoded in Configurator, proposed solution:

1. allow marking mixer as disabled
1. during initialization in mixerInit check if mixer is enabled. If not, fallback to default mixer setting (QuadX?)
1. cleanup specific code connected with disabled mixers

DzikuVx, 2016-04-02:
As a test I disabled and removed specific code to BiCopter mixer. 
@digitalentity please take a look at b4408a3eebb7bfaf6a3cfbbe372046d6894055df and badabd4d1e063297f3689449a2187263867ba0db
If approach is fine, I will continue. Plus, I've created wiki page with [removed mixers](https://github.com/iNavFlight/inav/wiki/Custom-mixes-for-exotic-setups)

No, there is still a question which mixers should go away. I vote:

* MIXER_BICOPTER -> done
* MIXER_QUADP anybody really usues Quad+ config? 
* MIXER_HELI_120_CCPM
* MIXER_HELI_90_DEG
* MIXER_VTAIL4
* MIXER_DUALCOPTER
* MIXER_SINGLECOPTER
* MIXER_ATAIL4
* MIXER_HEX6H -> That one puzzles me the most

digitalentity, 2016-04-02:
Initial implementation looks good and clean.

I'm not sure if some exotic mixers can be implemented using custom mixer. MIXER_CUSTOM does not support servos, MIXER_CUSTOM_TRI supports only one servo. If we blindly remove mixers we might break compatibility with rotorcrafts with more than one servo.

DzikuVx, 2016-04-03:
Kokos like you are right with limitations of custom mixers. I will dive deeper 😁

DzikuVx, 2016-04-03:
Yes, you are right. Current logic of choosing if to process servos and write servos depends not on actual rules, but mixer name.In this case, Change of plans. I will restore BiCopter, and cleanup only those mixers that are not using servos, or are not actually implemented. For example, both Helis. 

Next step would be to refactor how mixer handles servos. But that would be next task

digitalentity, 2016-04-03:
Don't focus too much on servo handling in this PR, let's make ability to compile out some mixers working first.

BTW, I suggest we don't really remove the mixers, but enclose them in **#ifdef** section - F3 targets have enough memory to support all mixers (even the exotic ones)

DzikuVx, 2016-04-03:
No no, no servo changes in this ticket. I still have to remember that I have compiler level directives. 18 years break with C after all...

closed 2016-04-03 by DzikuVx