iNavFlight/inav #141: Refactor how mixer handles servos

DzikuVx, 2016-04-03:
This is a continuation of #139

Currently, decision if to process servos depends not on stored servo rules, but type of a mixer. This makes custom servo mixer almost useless, since you can not enter all possible motor-mixer flying configurations

Tasks
* Always load custom servo mix in case on custom mixers
* Process servos when rules are defined, not based on mixer itself
* Build dynamic mapping table based on entered cutom servo rules 
* Write servos when servo is defined, remove hardcoded servo count limit, use mapping table prepared in previous point
* Cleanup remaining predefined mixers from #139

digitalentity, 2016-04-03:
Looks like a plan! Finally we'll have mixer code working in a clean and extendible way,

DzikuVx, 2016-04-05:
Summary of second day of this refactoring is: the things I've seen can not be unseen.
After digging into servo mixing and PWM mapping here what I've found:

* There is a global switch on PWM layer level that enables more than 2 servos only for MIXER_AIRPLANE, MIXER_FLYING_WING, MIXER_CUSTOM_AIRPLANE. It's called airplane mode, do not mix with ENABLE_STATE(FIXED_WING). It's different thing
* There can be only 2 motors in airplane mode and 8 servos
* On any other mixer than above its reversed: 8 motors and 2 servos
* So, on any mixer you can not have 3 motors and 3 servos. Mixer can compute that, but it will not be written to outputs
* Than makes Singlecopter and MIXER_BICOPTER mixer not functional. EVER. It's just not working
* The same goes for MIXER_GIMBAL -> there is not code to drive it
* Probably noone ever checked Dualcopter
* The same for MIXER_PPM_TO_SERVO -> there is not code for it so I disabled it
* I've refactored servo mixer so it's way more generic, but still, it's either 2 motors 8 servos or 8 servos 2 motors
* MIXER_CUSTOM now loads servo rules and processes them, but can write only 2 servos to PWM outputs (8 motors). And does not enable  ENABLE_STATE(FIXED_WING)
* MIXER_CUSTOM_AIRPLANE supports up to 8 servos (2 motors) and enables ENABLE_STATE(FIXED_WING)

So, @digitalentity here are questions/ things to decide: 
* without really going deep deep into hardware MIXER_SINGLECOPTER and MIXER_BICOPTER will not work and ***was never working***. I suggest disabling and removing completly for all targets
* MIXER_DUALCOPTER can now be implemented as MIXER_CUSTOM since it meets the limit of motors and servos. We can leave it, we can remove it, it does not matter

Overall flash size reduction is not big, but I think code is more "normal" and logical since it does not uses all those switches. 
I will do some more tests and then create a PR

digitalentity, 2016-04-06:
Yes, there is hardware limitations to number of concurrent motors/servos. 
I think MIXER_SINGLECOPTER & MIXER_BICOPTER can be removed completely.

DzikuVx, 2016-04-06:
I even checked how it was implemented in MultiWii. Even there it never went outside _beta_. So I will remove them and continue testing

Ralf-W, 2016-04-07:
@DzikuVx 
Wow! However I still do not understand why BICOPTER will not work. Like DUOCOPTER it needs 2 motors and 2 servos - but 4 servo rules to mix roll and pitch for these 2 servos.

Sorry being too inexperienced to be able to really support your excellent work.

DzikuVx, 2016-04-07:
Duocopter and bicopter might work. On my testbench servos move. But no way to really check it. 
But singlecopter would not work. It nerda 4 servos and two motors in non-fixedwing setup. And this is impossible right now. 
Improved custom mixer will allow to enter more setups, but predefined mixers will go away

Twilight, 2016-04-08:
proposition: all mixers should be treated equally, predefined mixers should be implemented in configurator instead of firmware.

DzikuVx, 2016-04-08:
@Twilight although this sound promising, it would not be compatible with Cleanflight Configurator. If something like this happens, it would have to begin from CF and CF Configurator

digitalentity, 2016-04-08:
@Twilight this is not possible. There is a hardware limitation on many boards that restricts what combination of servos and motors is possible. Usually it's 2 motors + 4 servos or vice-versa.

DzikuVx, 2016-04-08:
@digitalentity I have and idea how to actually fix that pwm mapping problem. But this would go deep deep into hardware issues and I'm not sure it's worth the effort. 
But one thing has to be fixed: docs on servo outputs and servo slot names. If I'm correct ATM servo slot name has no meaning right now since servos and motors are written to outputs in sequence order, not to predefined outputs like MultiWii does

digitalentity, 2016-04-08:
@DzikuVx it's limited by what hardware timers are used to drive outputs. Usually it timer is shared between i.e. output 1 and 2 they both can be either servos or motors. I'm sure something can be done on low level, but I'm also not sure it's worth the effort.
If some day we decide to heavily refactor pwm mapping code this can be fixed as well, but until then we'll have to cope with what we have.

rustyoz, 2016-04-10:
Hmm this sounds quite annoying. I was planning to run a custom quad with cleanflight on a spracingf3 with each axis having a tilt setup like those found on a tricopter, which will require 4 motor channels and 4 servo channels.

DzikuVx, 2016-04-10:
@digitalentity @rustyoz I really have a feeling that it's simpler that it looks. I think I understand pwm_mapping.c and if time allows I will try smth out. Who knows...

rustyoz, 2016-04-11:
so from what i understand pwm_mapping have constant structs which define 4 different PWM channel configurations. These definitions are hardware based and relate to the timers available. 

    multiPWM,
    multiPPM,
    airPWM,
    airPPM,

for example 

     static const uint16_t airPWM[] = {
     PWM2  | (MAP_TO_MOTOR_OUTPUT  << 8), // motor #1
     PWM3  | (MAP_TO_MOTOR_OUTPUT  << 8), // motor #2
     PWM4  | (MAP_TO_SERVO_OUTPUT  << 8), // servo #1
     PWM5  | (MAP_TO_SERVO_OUTPUT  << 8),
     PWM6  | (MAP_TO_SERVO_OUTPUT  << 8),
     PWM7  | (MAP_TO_SERVO_OUTPUT  << 8),
     PWM8  | (MAP_TO_SERVO_OUTPUT  << 8),
     PWM9  | (MAP_TO_SERVO_OUTPUT  << 8),
     PWM10 | (MAP_TO_SERVO_OUTPUT  << 8),
     PWM11 | (MAP_TO_SERVO_OUTPUT  << 8), // servo #8
     0xFFFF
    };

the selected mapping is determined in the function pwmInit() which selects from those 4 structs based on whether the receiver input is PWM or PPM and if the mixer is for an aeroplane. 

     if (init->airplane)
        i = 2; // switch to air hardware config
    if (init->usePPM || init->useSerialRx)
        i++; // next index is for PPM
https://github.com/cleanflight/cleanflight/blob/master/src/main/drivers/pwm_mapping.c#L657

This is set based on the current mixer mode set in the init function of main.c

    // when using airplane/wing mixer, servo/motor outputs are remapped
      if (mixerConfig()->mixerMode == MIXER_AIRPLANE || mixerConfig()->mixerMode == MIXER_FLYING_WING || mixerConfig()->mixerMode == MIXER_CUSTOM_AIRPLANE)
        pwm_params.airplane = true;
    else
        pwm_params.airplane = false;
https://github.com/cleanflight/cleanflight/blob/master/src/main/main.c#L342

Going forward I think it is possible to have hardware specific list of the PWM channels as related to the order of the pinouts on the board. Then the PWM init is done by first allocating a PPM or PWM channels, then the number of motors in the mix, then the number of servos. Where to start allocating motors and servos from the list of PWM channels would also be defined. This would be to match motor  and servo outputs with the physical pins on each specific board. 

The struct  drv_pwm_config_t in pwm_mapping.h and referred to as pwm_params in main.c would include the number of motors and servos from the mixer and be set after the mixer is loaded.

digitalentity, 2016-04-11:
The problem here is that we may have up to 4 outputs on a single timer. A timer has a common frequency divider meaning that all 4 channels must share the same refresh rate.
Brushless motors have 400Hz rate
Servos have 50Hz update rate

If brushless motor and servo share the same timer - it's impossible to use them.

rustyoz, 2016-04-11:
well in that case instead of different pwm configs based on the mixer mode ie multiPWM, multiPPM,airPWM,airPPM. 
we have different valid combinations of receiver, motors and servos. The number of motors and servos in the mix then determines from these combinations which one is selected.
eg. pwmm4s4 for pwm 4 motors and 4 servos. 
 ppmm2s6 for ppm 2 motors and 6 servos etc.

DzikuVx, 2016-04-12:
@rustyoz this can be done. In theory. But the problem it that it would have to be done for all hardware targets. And that means all targets would have to be tested. If I'm correct, iNav team combined does not have all boards to test that. Too bad, since I second that such a change would be super cool. Maybe this can be raised in cleanflight? Maybe they have more boards to play with this.

rustyoz, 2016-04-12:
I think it could be done for all targets just through code inspection, if there was a good reference to determine which pwm channels are controlled by which timers. I still haven't found where this is stated in cleanflight.

digitalentity, 2016-04-13:
@rustyoz you are more than welcome to assist with this.
PWM channels mapping to actual timer hardware is declared in timer.c. I.e. for NAZE: https://github.com/iNavFlight/inav/blob/master/src/main/drivers/timer.c#L46

DzikuVx, 2017-07-05:
I have  no idea why this is still open.... Closing :)

closed 2017-07-05 by DzikuVx