iNavFlight/inav #154: Add support for additional mw-nav compatible MSP way point types

stronnag, 2016-04-10:
The following MSP / mw-nav compat options are not currently implemented in inav. Please consider them for future releases. Maintaining MSP/mw-nav compatibility where practical is useful as it enables existing ground station software to be used. See https://docs.google.com/document/d/16ZfS_qwc-rJeA7N5Tx0DA6wtgxl6HdGgaz-jE3lHBWs, summary follows:

- POSHOLD_TIME (P1 = time in seconds)
- SET_POI (lat, lon): craft always 'head in' to the SET_POI position
- JUMP - P1 = prior WP number, P2 = number of iterations (-1 until mission cancelled)
- SET_HEAD P1=heading (P1 = -1 cancels this mode and SET_POI mode)
- LAND (lat,lon,alt??)

RTH landing option is covered by #61

MichaelCWarren, 2016-12-15:
Is there any current plan to implement this in the near future?

digitalentity, 2016-12-17:
That's a low priority at the moment. What types are you interested in?

MichaelCWarren, 2016-12-17:
I'm considering writing a mission planning iOS app and was curious what the future of this feature is. The current waypoint types are fine, but it would be neat if we could implement the others...

Michael

digitalentity, 2016-12-17:
We will eventually implement more.

anbello, 2017-05-18:
These new types of waypoints could be useful for aerial photography & video application, in this way we could use our DIY quads to make things similar to those made with expensive DJI quads (maybe).

digitalentity, 2017-05-18:
@anbello yes, we'll eventually implement more waypoint types. This is planned but very low priority task

anbello, 2017-05-18:
@digitalentity thanks
SET_POI could be the more interesting

b14ckyy, 2018-02-18:
Any updates here? I build an areal photography drone for an friend last year and everything works great. The only thing he (and me too) is disappointed in, is the lack of possibility to make orbital shots of buildings and sights or just let the drone "wait" at a certain point to make some shots with the gimbal while just leaving the drone fly it's mission.

eziosoft, 2018-04-04:
@digitalentity I don't get it. In the navigation oriented firmware features that can enhance automatic navigation are "very low priority"

neagron, 2018-05-08:
yes it would be great to have all the above features then it will be better than the vector and iNav is for waypoint mission's,please O please

jimpriest, 2019-03-03:
Adding my vote to add these features.

lemarec, 2019-08-14:
+1 For these features

jimpriest, 2019-09-22:
Any reason this has no interest?  Seems like an important part of the 'nav' in INav :)

georgekucher, 2019-12-03:
I have a VERY BIG interest for this feature. For a looooong time. FollowMe+LookAtMe.
Please...

Jacky2k, 2020-02-09:
Hi everyone,
I'd like to heat up this discussion again. I am currently trying to implement waypoint hold time function. The problem I came across: For this action p1 is defined as the time and there is no definition for the speed. For normal waypoints speed is also in p1. I'd like to support speed, so the only option without breaking compatibility is using p2 for the speed instead, I guess. Is that the way I/we should go?

Personally if I would redefine that protocol I would put the wait time into p2 of a normal waypoint. There is basically no difference in handling these two types except for waiting after reaching the WP. But I guess that would break compatibility too much? And it would confuse users as they would still have the option to create a timed pos hold wp in mwp and other software.
I am open to suggestions/comments.

stronnag, 2020-02-09:
> I'd like to heat up this discussion again. I am currently trying to implement waypoint hold time function. The problem I came across: For this action p1 is defined as the time and there is no [...]

Seems like an expedient solution. Nice to see some movement here.

Jacky2k, 2020-02-09:
OK, here we go: https://github.com/Jacky2k/inav/tree/jacky2k_waypoint_hold_feature
This is my piece of software that *should* enable waypoint hold time. I will not be able to test this in the real world for at least one week. If someone else is willing to risk his/her drone, please go ahead. I tested a bit with fake GPS, but I would be careful.
The time is now encoded in p1, the speed in p2. If you get the latest mwp from the development branch, it will not convert the hold time wps into normal ones.

stronnag, 2020-02-09:
If this is going to end up as a pull request (which is the only way it will be merged), it should be developed against development, not master.

Jacky2k, 2020-02-10:
Yeah, I was told so. I will try to do both this evening.

Jacky2k, 2020-02-10:
I just tried to create a pull request against the current development branch but something went horribly wrong. I just asked for a pull of my branch, not the entire development branch against itself. I don't get it. Can someone who knows git tell me what to do? :D

stronnag, 2020-02-10:
You made the mistake that we've all made at one stage; you didn't select development in the final stage of creating the PR. That has now been fixed.

Jacky2k, 2020-02-10:
Thanks a lot for that!
Can/should I use the development branch when testing this feature? I guess this branch is generally not considered stable?

stronnag, 2020-02-10:
Yes, always use the development branch, it's the most stable :)

stronnag, 2020-02-10:
So, masochist that I am, I locally patched @Jacky2k's PH Timed into @giacomo892's JUMP repo. Missions with JUMP and PH_TIME can be uploaded and downloaded via mwp and validate as the same mission. So, assuming this time I actually take the right `plane to the field, Wednesday could be interesting. Actually, I'm sure it will be interesting, just possibly not in the way I expect.

Jacky2k, 2020-02-10:
Don't say afterwards that you have not been warned :D

closed 2020-06-23 by stronnag