iNavFlight/inav #156: Waypoints, RTH: make heading change smooth

DzikuVx, 2016-04-10:
Currently, when switching to next waypoint or changing heading during RTH before descend, yaw input is very rapid. Specially when multirotor tries to make 180 degrees turn. It looks more less like this:

[image]

In case of underpowered machines with low yaw authority it leads to roll/pitch disturbance. It would be nice to have this yaw transition "smooth".

theArchLadder, 2016-04-10:
I agree with this. Feels like some jerk limit would be nice here, i don't see a problem with high angular velocity though.

digitalentity, 2016-04-10:
@DzikuVx, @theArchLadder 
I'm going to refactor the way MAG controller works and make it part of FP-PID controller (similar to heading lock mode). It will have some filtering built in to reduce jerk.

theArchLadder, 2016-04-11:
@DzikuVx posted a nice video of WP mission today. There seems to be no yaw movement before switching to the next waypoint, like "reached the waypoint, oh crap i need to turn!"
How about starting to yaw towards the next waypoint when the multirotor starts slowing down for the current waypoint...? So when switching waypoint, the yaw maneuver should be halfway done.
https://www.youtube.com/watch?v=p4ggWjN4GSM

oleost, 2016-04-11:
Looks like okey behavior if you ask me?

If you want a smoother turning can you just use more waypoints?

theArchLadder, 2016-04-11:
@oleost not sure what the point of your comment is, 3 people here seems to think yaw behavior has room for improvement, are you against changing yaw behavior in the way we have talked about here?

I don't really want smoother turning, i want my quad to travel straight to the waypoint, then travel straight to the next waypoint. This issue is not about changing the direction of travel, it's only about changing the heading (what direction the quad is pointing, but not necessarily moving).

oleost, 2016-04-11:
@theArchLadder I just said it looked good on the video you posted. And are very comparable to other waypoints mission I seen on youtube with different flightcontroller. But then again, thats gimbal smoothed out, so in reality in may be worse than that video.

DzikuVx, 2016-04-11:
@theArchLadder @oleost Well, I'm author of that video. Yes, gimbal smoothed it, but quad was jerking every turn. It does not have enough power to do rapid yaw changes.
Look at traces on BBlog screenshot I posted. A lot was happening during that course change

digitalentity, 2016-04-11:
We can't yaw in advance. The position controller logic is very simple - it only knows about target position, it has no idea that there are more waypoints coming. I'll smooth out heading controller at the PID level - we'll have smooth yaw rotation during WP missions then.

DzikuVx, 2016-05-11:
solved by #195 , closing

closed 2016-05-11 by DzikuVx