iNavFlight/inav #157: Zero throttle during RTH disables i-term for yaw

theArchLadder, 2016-04-10:
During the landing phase of RTH, it looks like zero throttle with stick arming disables the i-term for yaw, so the heading will not be locked in.

On the other hand, with i-term active, i had problems with windup when my quad had landed, i guess the uneven terrain gave a small heading change on touchdown that could not be corrected with the landing legs on the ground. This windup made two of the motors spin up at high rpm.

Not sure what the best solution is, how about having i-term active but more constrained?

DzikuVx, 2016-04-10:
I tend to agree with this observation. During todays experiments with Yaw I term I ended up in a situation that cased loose of stability and crash. I have low yaw performance and when mixer was trying to satisfy yaw it had not enough power on other axises.
Either that, or prop broke in mid air...

stronnag, 2016-04-10:
I've noticed this too. if I throttle down during RTH land then yaw lock is lost, even with airmode on.

theArchLadder, 2016-04-10:
@DzikuVx betaflight 2.5.2 has this in the changelog:
"Added piroutte protection during hard yaw manouvres. Yaw should not dominate over roll and pitch in some scenarios"
Sounds like something that could be helpful for you...? Has this been implemented in iNav?

DzikuVx, 2016-04-10:
@theArchLadder No, it was not. Perhaps it's worth taking from betaflight. 
@digitalentity it was introduced via c9a35d278d97963ae82c03a57ec93f161bee94f3 but I don't think it would solve my problem

digitalentity, 2016-04-11:
@DzikuVx yes, it's definitely worth taking

digitalentity, 2016-04-11:
It would be better to prevent integration or even reset yaw integral if requested yaw rotation rate is high enough.

digitalentity, 2016-05-04:
@DzikuVx, @theArchLadder, @stronnag 
Can you confirm that the issue is still present with latest master? I believe #160 should fix it.

DzikuVx, 2016-05-04:
As soon as I will be able

theArchLadder, 2016-05-04:
I just remembered this on the last RTH i did just now. Based on this single test it did look like this issue is still present, when i cut the throttle the yaw started to wander and it had different behavior on the ground depending on throttle at zero or not.

Will hopefully do more tests later today.

digitalentity, 2016-05-04:
@theArchLadder, I've deliberately reduced PID authority at throttle below min-check (similar to way it's done in Betaflight). Maybe issue is this line: https://github.com/iNavFlight/inav/blob/master/src/main/flight/pid.c#L271

theArchLadder, 2016-05-04:
Sorry, i don't understand. This issue is about no i-gain when throttle is at zero, how is "reduced PID authority" related to this?

I think it would make sense to limit the authority of i-term so that we do not have so much windup/spool-up of the motors while on the ground, this seems like a separate issue, but perhaps this is what you meant?

digitalentity, 2016-05-04:
Heading lock overrides rcCommand[YAW] to maintain heading. Zero throttle + stick arming blocks rcCommand[YAW] at low throttle, so it's possible that the issue is in stick arming, not I-term itself

theArchLadder, 2016-05-04:
That does sound more likely, but why would the heading change then?

digitalentity, 2016-05-04:
Blackbox log will clarify the situation if you have one.

digitalentity, 2016-05-09:
I-term reset at zero throttle finally confirmed.

digitalentity, 2016-05-09:
Should be fixed now. I-term at zero throttle requires AIR MODE to be enabled.
This commit doesn't attenuate I-term at zero throttle and also fixes some bugs in anti-windup handling.

digitalentity, 2016-05-09:
Please test and re-open if still present.

theArchLadder, 2016-05-10:
I just tried RTH with zero throttle now.
It does change the heading at all
It goes up in altitude, but it does not care about altitude when flying home so it sort of slowly falls and crashes at the home location.

Works as expected with mid throttle though.
Not sure if i can re-open this issue myself...?

digitalentity, 2016-05-10:
@theArchLadder so it does change heading or it doesn't change heading? Also do you stick arm?

As for loosing altitude during RTH - interesting. This shouldn't happen - pilot should be locked out of altitude control at RTH. Can you open a separate issue for this phenomenon?

theArchLadder, 2016-05-10:
@digitalentity Sorry, it does not. Yes i use stick arm.

I will open a new issue in a moment.

digitalentity, 2016-05-10:
Stick arming completely disables Yaw control at zero throttle. This is going to be like this until we introduce new MAG HOLD controller as a part of fp-pid code.

DzikuVx, 2016-05-10:
On what I'm working currently BTW

theArchLadder, 2016-05-10:
It should still keep the current heading, right? I assume it behaves as if yaw stick was centered?

I have seen the heading change during the landing phase of RTH when throttle is at zero.

digitalentity, 2016-05-10:
@theArchLadder current MAG HOLD works by emulating Yaw stick input that's being ignored later by stick-arming code. What you are seing is probably arbitrary heading drift. MAG HOLD has no authority at zero throttle.
@DzikuVx is up to something better :wink:

theArchLadder, 2016-05-10:
@digitalentity Sorry but i still don't understand, there is no heading drift even in rate mode...? The I-term for the yaw-pid-controller prevents this.

theArchLadder, 2016-05-11:
I feel stupid now, i-term is disabled at zero throttle when using stick arming, even in rate mode... How about using airmode during RTH? Or perhaps that's already used...?

DzikuVx, 2016-05-31:
I had a chance to observe this last weekend. Then throttle is zeroed during landing, heading is no longer kept. Looks strange and little scary

theArchLadder, 2016-05-31:
@DzikuVx Did you use stick arming?
The roll and pitch also becomes unstable for me, i guess this is because rate I-term is disabled... Would it work to have airmode active with this?

This is actually also a problem in althold+poshold for me, My hover throttle stick is just at 25% so i usually bring the throttle stick to 0 if i want to descend, but at that point roll and pitch becomes unstable.

DzikuVx, 2016-05-31:
No, I use AUX switch. Unfortunately I do not remember if it was AIR MODE or not. Probably not

theArchLadder, 2016-06-02:
I did a quick test with airmode, but i didn't notice any improvement.

So this issue is not specific to yaw, and not stick arming either.
The rate I-term disabling with zero throttle seems to make copter unstable, but if we fix that we will have motor spool-up on the ground...

digitalentity, 2016-09-01:
I wonder if we should close this one

commit 43328c2f (2016-05-09): PID: Fixed I-term low authority at low throttle. Fixes #157
closed 2016-11-21 by digitalentity