iNavFlight/inav #159: Naze32 Rev5 won't take "board and Sensor"

rholt51, 2016-04-11:


digitalentity, 2016-04-11:
Please describe the issue. It's unclear what you mean by "won't take board and sensor".

rholt51, 2016-04-11:
Sorry about that first submission. My cat hit the enter key before I had finished. It should be updated now.

oleost, 2016-04-11:
https://github.com/iNavFlight/inav/wiki/Advanced-accelerometer-calibration

You have to degress x 10 and do it in CLI.

Also read rest of wjki while your at it ;)

This can be closed

DzikuVx, 2016-04-11:
@rholt51 iNav works with centidegrees and Configurator is not supporting this feature. To enter correct alignment you have to use CLI:
```set align_board_yaw=900
save```

closed 2016-04-11 by DzikuVx

digitalentity, 2016-04-11:
@oleost, @DzikuVx you are fast :+1:

rholt51, 2016-04-11:
Solved. Working now!