iNavFlight/inav #162: Wont arm in passtrough mode, fixed-wing.

oleost, 2016-04-14:
Compiled using latest master 13/04-16

Looks like there is a bug that prevents me for arming in passthrough mode. Arms in fine in acro mode.

Tried to do a blackbox log, but unsure how successfull it was. 
1. is in passtrough mode, and it wont arm/throttle, 
2. is in acro and it arms and throttles up.

[CLI dump, blackbox log and build used
](https://drive.google.com/open?id=0B-yHzNtOXxbERXFQZGd6MzBrcU0)

Can it have anything to do with [Extend extra arming safety - prevent arming when any of NAV modes is selected ](https://github.com/iNavFlight/inav/commit/8ac607c47c20e4730dbbbc6156c634c0049fd45c)?

That passtrough is wrongly marked as an nav mode?

digitalentity, 2016-04-14:
@oleost this is what causes your problem:
`set failsafe_procedure = RTH`
This setting may engage GPS-assisted mode, so it also causes extra_arming_safety to block arming until GPS is acquired and HOME position known.

oleost, 2016-04-14:
This is with GPS locked. Since I also have nav safety turned on.

But why would it make any difference armjng in acro mode compared to
passthrough mode?

digitalentity, 2016-04-14:
Oh crap, sorry!
That's a dump copy-paste error. In current code BOXPASSTHRU is indeed blocking arming. Fixing.

closed 2016-04-14 by digitalentity

oleost, 2016-04-14:
TY

commit 4a86f5d1 (2016-04-14): SAFETY: Fix BOXPASSTHRU blocking arming when extra_arming_safety is used. Fixes #162