iNavFlight/inav #163: Prolem with setting soft serial on SPRACINGF3

krzysztofmatula, 2016-04-14:
I'm not sure if this is inav problem, but I'm observing it here.
I wanted to set up soft serial and I typed in CLI:

feature SOFTSERIAL
<reboot>
serial 0 1 115200 57600 0 115200
serial 1 2 115200 57600 0 115200
serial 2 1 57600 57600 0 115200
serial 30 1 9600 38400 0 115200
serial 31 1 9600 38400 0 115200
save

Then, after reboot:
# serial
serial 0 1 115200 38400 0 115200
serial 1 0 115200 38400 0 115200
serial 2 0 115200 38400 0 115200
serial 30 0 115200 38400 0 115200
serial 31 0 115200 38400 0 115200

And all settings are gone...
Any Idea what may be wrong? I believe the setup I provide is OK for this platform.

Other features I use:
feature RX_PPM
feature VBAT
feature GPS
feature FAILSAFE
feature TELEMETRY
feature RSSI_ADC
feature LED_STRIP
feature ONESHOT125
feature BLACKBOX

DzikuVx, 2016-04-15:
If I remember correctly, order would be:
```
feature SOFTSERIAL
save 

serial 0 1 115200 57600 0 115200
serial 1 2 115200 57600 0 115200
serial 2 1 57600 57600 0 115200
serial 30 1 9600 38400 0 115200
serial 31 1 9600 38400 0 115200
save
```
You can not enable and configure a feature in the save "session"

krzysztofmatula, 2016-04-15:
You're right, I do it correctly. I wonder if anyone is actually using soft serials with this board... I'll check how it behaves on cleanflight head.

DzikuVx, 2016-04-15:
I'm using it for S.Port telemetry between SPRacingF3 and FrSky X8R. Works like a charm

DzikuVx, 2016-04-15:
OK, since it's working now, I'm closing iy

closed 2016-04-15 by DzikuVx

krzysztofmatula, 2016-04-17:
FYI, the problem was that I attempted to enable 3rd MSP instance... My fault.