iNavFlight/inav #164: gps reading wrong

richsemc, 2016-04-15:
I have a 250 quad with SPracing F3 board, with Crius CN-06 V3 Ublox GPS module.
The board is flashed with Inav and is setup with luxfloat. the compass is fully isolated from and interference , gps reads 10 sat's and shows position fine on mobile EZ-GUI, altitude measures fine when I take off. the quad flies perfect on Angle mode. and alt hold
The problem I have had for the past 4 weeks is that  gps  hold it starts flying of with it's nose down and looks as though it is going back to china in wild movements, changing all pid settings makes no difference. Pos, Posr, Mag, NavR
I have used blackbox tools and converted files to csv,gpx, when I look at the co-ordinance on google maps it shows me 49km east of my position, my home position is 53.38771880 -0.370416800 it looks as though it's not taking into account the minus on the software but EZgui does
links to attached files
https://www.dropbox.com/s/fzalxaj5sqlvnw0/bblog154.01.csv?dl=0
https://www.dropbox.com/s/1kgmr0taovigxyg/bblog154.01.event?dl=0
https://www.dropbox.com/s/fu3mvg48pbvsgve/bblog154.01.gps.csv?dl=0
https://www.dropbox.com/s/t4ebitw3ina0zbs/bblog154.01.gps.gpx?dl=0
full blackbox log file
https://www.dropbox.com/s/intoj0xaq9vx4a0/bblog154.TXT?dl=0
Richard welch

stronnag, 2016-04-15:
GPS positions are BB logged using the BB predictor from the home position. I wonder if there is some problem with this where the home position is near to the prime meridian. Note that Ezgui (which is working on direct (MSP/LTM) data displays the correct position.

richsemc, 2016-04-15:
My CLI dump file
https://www.dropbox.com/s/s6agc4iu9s29sx8/clidump154.txt?dl=0

digitalentity, 2016-04-16:
Thanks for reporting. I'll check if there is a bug with GPS coordinate conversion. INAV works in local coordinates (GPS degrees is converted to local coordinates in [cm] from point of first GPS fix. I wonder if the conversion doesn't work good enough near the prime meridian.

stronnag, 2016-04-16:
As the position is reported as correct in EzGUI (MSP or LTM), perhaps this is just to do with the way the predictor is used in bbox?. @richsemc, which protocol are you using with ezgui (MSP or LTM)? Can you provide a screenshot of the data in Ezgui?

richsemc, 2016-04-16:
I have downloaded the EZ-GUI files, the raw data from EZ-GUI and the converted CSV file which shows up perfect location on my mobile App
https://www.dropbox.com/s/j5mqh6nslxxbeau/EZGUI%20log15042016.csv?dl=0
https://www.dropbox.com/s/7802d7i801wjjuc/Flight_2016_04_15_11_26_00.el2?dl=0

richsemc, 2016-05-25:
any news on the above problem please

richsemc, 2016-08-02:
can someone please look at problem and advise

stronnag, 2016-08-02:
Can you try the latest master (noting the PID  and rate changes from the front wiki page https://github.com/iNavFlight/inav/wiki). If you need a hex built for you, please let me know (by adding a new comment here).

Ralf-W, 2016-08-02:
@richsemc 
Just some questions about your cli dump: 

Is your board alignment set correctly?
set align_board_roll = -44 # this means -4.4 degrees, not -44 degrees. 
set align_board_pitch = 4 # this means +0.4 degrees, not 4 degrees. 

Are your calibrations correct?
set magzero_x = 388
set magzero_y = 55
set magzero_z = 206
set acczero_x = 113
set acczero_y = -11
set acczero_z = -668 # value seems unusually high to me
set accgain_x = 4069
set accgain_y = 4103 # I have never seen this value above 4096 before
set accgain_z = 4023

set mag_declination = 0 # Is this correct close to the Meridian?

set gps_nav_model = LOW_G # present default value is AIR_1G - You should change it, but I do not expect this setting to affect your present problem.

digitalentity, 2016-08-03:
@richsemc 
Looks very much like compass issue to me.
Very hard to troubleshoot as you are always hovering. Can you please do a flight of at least 5-6 seconds using only pitch to fly forward and record a new log? No need to engage PosHold, the key here is to fly in **forward** direction in a straight line without using roll or yaw.

digitalentity, 2016-08-03:
On the other hand - no need to test. You have extreme interference on your compass from power wires, ESCs, motors, PDB or battery.
[image]
In the picture you can clearly see the correlation between throttle stick (below) and magnetometer readings (above). In all your flights I'm seing heading or ~20deg when throttle up and ~35 deg when throttle down.
Can you upload a photo of your setup and a close-up of your compass placement?

richsemc, 2016-08-08:
Thanks for your comments, I will try all above and come back to you

digitalentity, 2016-09-01:
@richsemc any news?

richsemc, 2016-09-01:
Hi
I have not had much time to look into it further, I have cured the compass problems so I have no interference but have not had time to do the rest, we had a house fire in a rented property and have got a lot of work to do on it, so I will get back onto it as soon as possible. and let you know, cheers ,richard

digitalentity, 2016-09-01:
Sorry to hear about the fire. I hope you got everything sorted out! Good luck!

digitalentity, 2016-11-21:
Closing. Looks like a blackbox decoder bug.

closed 2016-11-21 by digitalentity