iNavFlight/inav #165: Prefer GPS heading on airplanes

digitalentity, 2016-04-17:
A-ha! So for airplane it's better to use GPS heading (this is what INAV does on airplanes when there's no mag available).

However GPS heading is not always available/accurate. I think of airplanes INAV should just prefer GPS heading when available and fallback to MAG is there is a compass chip.

stinkydiver73, 2016-04-25:
What about stronger wind? Is it possible the plane will fly backwards to the way-point without realizing it. GPS heading is interesting, but we must never trust it over the MAG.

oleost, 2016-04-25:
With the #168 the GPS heading should always be accurate, anyway if your airplane is flying backwards you got other issues.

stinkydiver73, 2016-04-25:
accurate != optimal. If windy, you can fly A to B on a straight line with very different real (MAG) heading (with different air-speed), but always the same GPS heading. Which real heading you will fly, or who cares? I don't know the controller will reach some optimal heading or can stuck on some far from optimal one.

oleost, 2016-04-25:
http://www.rcgroups.com/forums/showpost.php?p=34540247&postcount=5126. Notice difference between with and without mag.

Without mag the planes flies sideways but directly to waypoint(home)

With mag it flies in a oval line towards waypoint but always keeping it pointed towards waypoint, when it should be turning more to fight wind.

digitalentity, 2016-04-25:
@stinkydiver73 
Using GPS heading has a side effect of inherently compensating for cross-track error (this is what @oleost is describing). As INAV doesn't have a cross-track correction code, if we use mag for heading we would point airplane directly to target, but this wouldn't mean that we would fly directly to target.

The solution would be to modify IMU and use GPS heading when it correlates with magnetic heading and fall back to magnetic heading otherwise. This will ensure that we don't use GPS heading when flying tail first in a strong headwind.

digitalentity, 2016-04-25:
A cleaner solution would be to calculate and compensate for sidewind and correctly calculate and account for crosstrack error. However I doubt INAV will ever have that - it is intended to keep things as simple as possible.

stinkydiver73, 2016-04-25:
Yes, i understand why the GPS heading can be "better". I just want to "force" to make the MAG handling better, not to drop it. All these valid for copters too.

digitalentity, 2016-04-25:
Yes, I see your point and I agree that there are moments when mag heading is better - we need to account for that for sure.
BTW: multicopters don't use GPS heading.

digitalentity, 2016-04-25:
Discussion should continue in the PR #172

commit 72e03c7e (2016-04-26): IMU,FW: Fallback to MAG if GPS CoG is significatly different from estimated heading. If GPS is the only source of heading it will be used regardless. Closes #165
commit 2959a8cc (2016-11-03): IMU,FW: Fallback to MAG if GPS CoG is significatly different from estimated heading. If GPS is the only source of heading it will be used regardless. Closes #165
closed 2016-04-25 by digitalentity