iNavFlight/inav #174: HW - Support I2C-to-PWM chip

digitalentity, 2016-04-27:
For those who want lots and lots of servos on their machines - https://www.adafruit.com/product/815

DzikuVx, 2016-04-27:
This actually is an excellent idea! The only problem is that to really implement this in a nice way some refactoring of pwm_mapping would be required. 
On the other hand, if it would work like: "only FC servos" / "only external servos" that might end up only with new ifs. 

There is also a question: this fancy piece of hardware or open-sourced Arduino mini code for 1/5 the price and only slightly less functionality?

digitalentity, 2016-04-27:
I feel that PCA9685 will be more reliable than Arduino board. If this thing is going to be used to drive servos on an actual aircraft, reliability should be a primary concern.

Yes, pwm_mapping will have to be totally reworked to support not only inbuilt timers but also external devices. This might be benificial - think about direct connectivity to ESCs via I2C or CAN - such ESCs are already available on the market (although very rare)

DzikuVx, 2016-04-27:
C'mon, people were flying MultiWii for years :)

About pwm_mapping. I do not think complete rework is needed  just to make this work. The way I see it:

1. detect I2C iterface and in pwmInit always force useServos to false. This way no FC timers would be used for servos
1. Instead initiate pwmOutputConfiguration.servoCount and pwmOutputConfiguration.outputCount with numbers from mixer, not hardware. Of course clip it to external device capabilities
1. Use pwmWriteServo as a facade, not implementation and write to either FC or external servo interface

This is was simpler than rework of pwm_mapping.c and actaully doable without retesting everything on any available target.
I'm currently quite short on time, but this is intresting enough to invest some time in it

digitalentity, 2016-04-27:
PCA9685 boards are as cheap as Arduino's on eBay - i.e. this one http://www.ebay.com/itm/-/201402794566

If you are willing to invest some time into refactoring the pwm_mapping, I'll start writing a driver for PCA9685 and send you a board to test on your end :smiley:

DzikuVx, 2016-04-27:
I do not want to refactor pwm_mapping, but I want to implement that board ;)
Let's do it like this: I will get me PCA9685 from ebay auction myslef since that might be useful in different projects. If you have time, write a driver. If you do not, then do not, I think my skills are high enough to port adafruit version here. Either way, when I will get my PCA9685 I will write iNav implementation. 
And before that I should have enough time to finally finish my battery voltage as virtual AUX

digitalentity, 2016-04-27:
@DzikuVx 
I have a PCA9685 board somewhere in my desk. I'll give it a go and spend some time writing a generic driver for INAV. I still have a feeling that pwm_mapping will have to be heavily refactored in near future...

DzikuVx, 2016-04-27:
The problem with pmw_mapping is that CF supports too many different boards

digitalentity, 2016-04-27:
I think PWM layer should be made target-dependent. We'll have another file in target/BOARD directory - it will abstract PWM initialisation and handling and present a writeOnboardMotor() writeOnboardServo() functions.

DzikuVx, 2016-04-27:
I think you are going too far with this ticket now. Support for PCA9685 is one, complete refactoring of pwm_mapping is different

digitalentity, 2016-04-27:
Yes, let's keep this on topic.

DzikuVx, 2016-05-17:
I just got my PCA9685 today. Ready to proceed :)

DzikuVx, 2016-09-15:
#261 merged, closing

closed 2016-09-15 by DzikuVx