iNavFlight/inav #176: Looptime <2000 doesn't work on F3 + Gyrosync not work

dyter, 2016-04-28:
Looptime configuration is saved but not in cpu status

gyrosync=ON doesn't work (can not use more cpu)

[image]

digitalentity, 2016-04-28:
Unable to reproduce the issue on SPRacingF3:
[image]

digitalentity, 2016-04-28:
What board do you have?

dyter, 2016-04-29:
It's a Beerotor F3

digitalentity, 2016-04-29:
Just to verify - did you power-cycle the board after enabling gyro sync?

dyter, 2016-04-29:
I set looptime to 250 but it's stay to 1000, I don't understand !

[image]

digitalentity, 2016-04-29:
As I said in #175":
> `gyro_sync=ON` makes `looptime` setting irrelevant.

```
gyro_sync=ON
gyro_sync_denom=1
```
sets the cycle time to 1000. It is unwise to attempt to go lower than that on INAV.

digitalentity, 2016-04-29:
Apparently gyro_sync works on your board. Issue is closed.

closed 2016-04-29 by digitalentity