iNavFlight/inav #177: [CC3D] SoftwareSerial conflict with servo outputs airplane

sppnk, 2016-04-29:
Tested in 1.1 RC1 and 1.1 RC2 releases and three different CC3D boards.

If I use feature SOFTSERIAL (I have tried to use telemetry), the output 5 (aileron) in CC3D stops working, and the output 6 (rudder) behaves like an aileron.

stinkydiver73, 2016-04-29:
It's a known hw conflict.

sppnk, 2016-04-29:
@stinkydiver73, I see you have solved it through self compilation code ([here](http://www.rcgroups.com/forums/showpost.php?p=34482982&postcount=4865)), so, if it can be solved with software, why do say it is a hardware issue?

digitalentity, 2016-04-29:
Yes, this is a known issue. Can't be fixed at the moment unfortunately.
As a temporary workaround for the issue I'll disable SoftSerial when AIRPLANE mixer is used.

sppnk, 2016-04-29:
ok, thks @digitalentity . I would like to know if it is a hardware conflict (no solution) or a code problem that could be solved in the future.

digitalentity, 2016-04-29:
Please keep the issue open. I might be able to create a workaround for this.

digitalentity, 2016-04-29:
It's a hardware conflict - SoftSerial on CC3D uses TIM3 timer, same timer is used for Output 5. However it does work on multirotor crafts, so I suspect we can have a (hacky) workaround for airplanes.

digitalentity, 2016-04-29:
Ok, it doesn't work for multicopters as well. You can't have a Hexacopter + softserial on CC3D :disappointed:

digitalentity, 2016-04-29:
Timers are allocated in such way that we can have only 6 output OR softserial on CC3D. As far as I can tell this is hardware conflict that can't be resolved. #174 might be a bulletproof solution for CC3D airplane setups.

digitalentity, 2016-04-29:
As a workaround we may disable Output 5 completely (at all setups) and limit the motor outputs to one. 
This way we will have this setup regardless of SoftSerial feature.
Out1 - Motor
Out2 - Elevator
Out3 - Aileron
Out4 - Aileron
Out5 - not used
Out6 - Rudder

Won't solve the problem for hexacopters though.

sppnk, 2016-04-29:
[Here](http://www.rcgroups.com/forums/showpost.php?p=34482982&postcount=4865) @stinkydiver73 made something similar I guess

digitalentity, 2016-04-29:
@sppnk yes, but his solution still moves Rudder between 5 and 6 depending on softserial being disabled or enabled. I suggest we drop two motor outputs on CC3D altogether.

digitalentity, 2016-04-29:
It's a very hacky and ugly solution, it will make motor mixer platform-dependent which is not a good thing. We are in need of complete pwm_mapping thing rework.

sppnk, 2016-04-29:
Ok, good, in the meanwhile I will try to use USART1 for bluetooth(MSP EZGUI configurations) + telemetry (testing now Frsky, but going to LTM solution). I will ask for this in rcgroups.

digitalentity, 2016-04-29:
I use similar thing on my Ranger. UART1 is MSP (bluetooth) + LTM (OSD). UART2 is GPS. Bluetooth connection to EzGUI works when disarmed, LTM and OSD works when armed.

sppnk, 2016-04-29:
That is great. I have read somewhere that EZGUI can understand LTM packets. I am planning to have the telemetry downlink to my phone, and LTM is a perfect compact simple protocol for iNav infos.

sppnk, 2016-04-29:
@digitalentity , no sure, but if we use only one output for ailerons ('Y' cable) we would have another free output without TIM3 conflict. No posibility for flaperons mix, but could be a better solution. Anyway, in automated flying planes I dont see very important the flapperons mix, airbrakes, etc...what do you think?

stinkydiver73, 2016-04-29:
A more flexible mixer would be good, with no fixed number of motor output. Or another custom airplane mixer with 1 motor. I don't understand why motor and servo output differ? Because of the different refressh rate?
(Off topic, more serial port sharing options, or just better boards with more hw serial)

digitalentity, 2016-04-29:
@stinkydiver73 it's different refresh rate. Motor outputs have 490Hz or even variable refresh rate (when in OneShot mode). Servos are 50Hz.

DzikuVx, 2016-04-29:
@sppnk I'm not so sure of this MPS/LTM switch in EZGUI. I failed to set it up last weekend.

sppnk, 2016-10-16:
Hi, I want to use softserial in my CC3D boards to have a downlink of LTM telemetry throught my openlrsng receiver to my EZGUI app at groundstation.

It seems that softserial is available in the code, but I am not sure if this issue was solved or not.

If so, please confirm and close this issue. If not, softserial shouldnt be available for CC3D. 

Can anybody take a look to this? Thkx.

digitalentity, 2016-10-16:
@sppnk there is a timer clash between SoftSerial and OUT5. Unfortunately there's no software solution available. 
At the moment SoftSerial is usable on mixers that don't use Out5 (Quad, Tri, Flying wing etc)

digitalentity, 2016-10-16:
I'm not sure that it's a bug. It's more of hardware limitation.

sppnk, 2016-10-16:
ok. I understand. So in a plane it is not usable.

 Should it be possible then to use Out2  just for softserial TX? Motor 2 is not used.

This way I could have a LTM downlink to my EZGUI phone app (I cant do that for now using USART1, because I am using MSOSD at the same time, and it needs MSP).

digitalentity, 2016-10-16:
@sppnk i'm afraid not. We don't have a timer that doesn't clash with either PPM/PWM input or motor output and is available on a spare pin.

sppnk, 2016-10-16:
ok, I will try to use MSP tx line from osd USART to send it to ground. Otherwise I should switch to a better FC with more ports available. thks.

digitalentity, 2016-11-21:
Hardware limitation.

closed 2016-11-21 by digitalentity