iNavFlight/inav #178: Vfas support

MikeBland, 2016-04-30:
In your "Telemetry.md" file you state "VFAS resolution of 0.1 volts and is only supported by OpenTX".
Please note ID 0x39 was added to support the openXsensor and this ID is fully supported by ersky9x firmware on the Taranis, 9XR-PRO, 9Xtreme, AR9X and SKY platforms.

DzikuVx, 2016-04-30:
@MikeBland would you mind creating a Pull Request to docs? Looks like you know what you write :)

MikeBland, 2016-04-30:
To do that I think I'd have to clone the repository first, wouldn't I?
All that needs changing is this:
"This is new setting which supports VFAS resolution of 0.1 volts and is only supported by OpenTX radios (this method uses custom ID 0x39)."
to something like this:
"This is new setting which supports VFAS resolution of 0.1 volts and is supported by OpenTX and er9x/ersky9x firmware (this method uses custom ID 0x39)."

Thanks.

DzikuVx, 2016-05-01:
In this case, docs updated, ticket closed. 
Thank you very much for suggesting this change!

closed 2016-05-01 by DzikuVx

digitalentity, 2016-05-01:
@DzikuVx, @MikeBland thanks!