iNavFlight/inav #183: LTM - Allow lower baudrates than 9600bps

sppnk, 2016-05-03:
AFAIK, when baudrate decreases, the range increases. So It would be good to permit lower baudrates to LTM protocol (until 1200bps perhaps).May be for a audio-over-video downlink or other telemetry systems.

I my actual setup I use openlrsng, and I guess that the 'serial baudrate' for it has to be lower than 'baudrate' (radio connection). So if I use 'baudrate' = 9600 bps, then there is no margin for 'serial baudrate' lower than that. Would be nice if I could set LTM to 2400 and radiolink to 9600. For airplanes in mission WP could be down to 4800 bps.

Now I have set radiolink ('baudrate' in openlrsng) to 19200 and 'serial baudrate' to 9600 in order this to work. And it does, but I would prefer to have more range.

Please, feel free to correct me if I am wrong with something of this.

DzikuVx, 2016-05-04:
I can only agree. Lower baud rates would be nice addition. I wonder if min at 9600 is some kind of legacy from MultiWii or there is a real reason for it?

digitalentity, 2016-05-04:
I've looked into it and if we do it we will break compatibility with Cleanflight Configurator as it operates indexes, not numeric baud rates.
What we can do is force AUTO baud rate for LTM telemetry to be actually 1200 baud.

stronnag, 2016-05-04:
Current LTM is effectively at or perhaps just over 2400baud. The original implementation catered for low (<2400) and higher (>2400) rates. This was removed for CF due to 9600 minimum. We'd need to reimplement for 1200, which would be easy.

digitalentity, 2016-05-04:
@stronnag 
LTM for low and high baud rates would be a good thing!

Once I release initial version of INAV configurator I intend to deliberately break compatibility with original Cleanflight configurator to avoid issues with board alignment etc, We can look into adding lower baud rates to Configurator then.

What do you think?

stronnag, 2016-05-04:
Yes to both

digitalentity, 2016-07-21:
Scheduling for 1.3 to be done along with #365

closed 2017-03-27 by digitalentity