iNavFlight/inav #185: Using RTH when no GPS fix available causes fast descent

iforce2d, 2016-05-05:
My quad has been having trouble keeping a GPS lock during flight. In position hold mode, when this happens it will start drifting with the wind. If I switch into RTH mode at this time, the quad will immediately descend VERY quickly, almost at falling speed, as if the motors are just idling.

I have position hold and RTH on the same switch, so the mode change is going from position hold to RTH mode, if it makes any difference. That is, I have not tried going from manual to RTH modes.

So far I have not seen what happens when GPS lock is lost during a currently running RTH sequence.

My expectation was that even if there is no GPS fix, RTH could still do the initial ascending phase, or at the very least should preserve the current altitude.

digitalentity, 2016-05-05:
If no GPS available at the moment you switch to RTH the quad will start emergency descent at 5 m/s (default value of `nav_emerg_landing_speed` setting. This is intended as a safest way to prevent fly-aways.

Imagine RTH triggered by failsafe. You don't have control of your aircraft and you won't have any chance of recovering it if it starts ascent phase or will maintain altitude and drift with the wind until battery dies.

iforce2d, 2016-05-05:
I see. It's just so fast that it looked to me as unintended behavior. Is there a CLI variable to adjust the speed?

digitalentity, 2016-05-05:
The setting is `nav_emerg_landing_speed`.

digitalentity, 2016-05-05:
I'm reopening this issue as it's actually a bug - the firmware should wait for `nav_position_timeout` since last valid GPS fix before initiating emergency landing.

iforce2d, 2016-05-05:
In the cases I tried, the timeout since last valid GPS fix would have already been exceeded during position hold. Do you mean that the timeout should be reset when entering RTH? In that case yes, it would be a bug.

digitalentity, 2016-05-05:
No, firmware should check if GPS was lost a millisecond before RTH activation and wait for timeout (5 seconds default). The code doesn't do it - it enters emergency descent immediately.

digitalentity, 2016-05-05:
Verified that firmware actually **does check** for GPS timeout and switch do emergency landing only after that.
Immediate transition to emergency landing will happen if timeout has already elapsed when RTH is activated or if home position is not set (i.e. drone took off without GPS FIX and with disabled extra arming safety).

closed 2016-05-05 by digitalentity

iforce2d, 2016-05-05:
Thanks again. Slightly beside the point, but would setting nav_emerg_landing_speed to zero effectively disable this landing behavior? Or is there a better way to do it?

The reason I ask is for situations like mine, where I am more worried about losing GPS lock than losing radio control.

digitalentity, 2016-05-05:
You can do `set nav_position_timeout = 0` to disable the timeout. GPS will never time out and quad won't switch to emergency landing.

digitalentity, 2016-05-05:
Actually quad still may switch to emergency landing on certain occations - like invoking RTH without HOME position.