iNavFlight/inav #187: Drift in roll with still FC although ensors indicate no movement.

sppnk, 2016-05-07:
Hi, I am having the same issue like #1780 and #2079 in cleanflight. It is an afromini naze32.

Having the plane completly stopped on the bench and connected to configurator, unarmed  I see how slowly the roll indicator on setup tab drifts. The sensor tabs indicates no movement. When resetting, the values of roll anf pitch go to zero  and then start to move again. If I press acc calibrate on setup tab it doesnt reset attitude values.

digitalentity, 2016-05-07:
Please post a screenshot of your sensor tab.

sppnk, 2016-05-07:
[image]

This is sensors tab,

And this is setup tab after 30 secs:

[image]

sppnk, 2016-05-07:
My complete dump for this FC:

`
# dump

# version
# INAV/NAZE 1.1.0 Apr 30 2016 / 12:28:16 (19c4c07)
# pflags
# Persistent config flags: 0x00000000
# dump master

# mixer
mixer AIRPLANE
mmix reset
smix reset

# feature
feature -RX_PPM
feature -VBAT
feature -INFLIGHT_ACC_CAL
feature -RX_SERIAL
feature -MOTOR_STOP
feature -SERVO_TILT
feature -SOFTSERIAL
feature -GPS
feature -FAILSAFE
feature -SONAR
feature -TELEMETRY
feature -CURRENT_METER
feature -3D
feature -RX_PARALLEL_PWM
feature -RX_MSP
feature -RSSI_ADC
feature -LED_STRIP
feature -DISPLAY
feature -ONESHOT125
feature -BLACKBOX
feature -CHANNEL_FORWARDING
feature RX_PPM
feature MOTOR_STOP
feature GPS
feature FAILSAFE

# beeper
beeper GYRO_CALIBRATED
beeper RX_LOST
beeper RX_LOST_LANDING
beeper DISARMING
beeper ARMING
beeper ARMING_GPS_FIX
beeper BAT_CRIT_LOW
beeper BAT_LOW
beeper GPS_STATUS
beeper RX_SET
beeper ACC_CALIBRATION
beeper ACC_CALIBRATION_FAIL
beeper READY_BEEP
beeper MULTI_BEEPS
beeper DISARM_REPEAT
beeper ARMED
beeper SYSTEM_INIT
beeper ON_USB

# map
map AETR1234

# serial
serial 0 1 115200 38400 9600 115200
serial 1 2 115200 38400 0 115200

[... 298 lines cut ...]
set i_level = 15
set d_level = 75
set max_angle_inclination_rll = 300
set max_angle_inclination_pit = 300
set gyro_soft_lpf_hz = 60
set acc_soft_lpf_hz = 15
set dterm_lpf_hz = 30

# dump rates

# rateprofile
rateprofile 1

set rc_rate = 90
set rc_expo = 70
set rc_yaw_expo = 20
set thr_mid = 50
set thr_expo = 0
set roll_rate = 0
set pitch_rate = 0
set yaw_rate = 0
set tpa_rate = 0
set tpa_breakpoint = 1500

# `

digitalentity, 2016-05-07:
Your accelerometer is not calibrated. Please do a 6-point accelerometer
calibration as described in the wiki.

sppnk, 2016-05-07:
Done and... no more drift, (just a little in heading, but I will recalibrate -difficult with the plane-) . It works. Sorry for take your time and I close this issue.

closed 2016-05-07 by sppnk