iNavFlight/inav #190: acc calibrate problem

enrique15241524, 2016-05-08:
hi;
i did load inav 1.1 c3
but it is not acc calibrate..
after calibrate values:
pitch=-1.3
roll = 0

regards
Murat

[image]

skaman82, 2016-05-08:
Have you read this? https://github.com/iNavFlight/inav/wiki/Advanced-accelerometer-calibration

enrique15241524, 2016-05-08:
thanks skaman :)
i bad english :)

i applied this steps and values:
is normal?

[image]

sppnk, 2016-05-08:
please check this : #187

skaman82, 2016-05-08:
@enrique15241524 yes, a small offset after calibration is normal. You can tune the small offset in the configuration tab in this area: 
[image]

enrique15241524, 2016-05-08:
Thanks for all help... 
i will try tomorrow 
so do not close the topic...

Regards....

theArchLadder, 2016-05-09:
I think this shows that this is not user friendly, i think most people expect the acc calibration to also do a level calibration. Not sure why this is not done... @digitalentity?

digitalentity, 2016-05-09:
@theArchLadder two reasons:
1) Calibration means only that sensor chip is providing consistent and sane data. With compass calibration doesn't mean you'll have your heading right - you need to adjust magnetic declination. Accel is alike - calibration doesn't mean it will read level correctly, it only means it will measure acceleration and gravity correctly.
2) We will have better calibration procedures - compass, accelerometer, level calibration in new Configurator (hopefully this will happen shortly after 1.1 release)

oleost, 2016-05-09:
@digitalentity I understand what your saying.

But isn't it possible that when doing the 1. posision of the accelelrometer process it also adjust "align_board_pitch" and "align_board_roll" so it will say its leveled?

digitalentity, 2016-05-09:
@oleost 1-st position isn't necessary level, it has margin of ~15 degrees. The 6-point calibration is designed in such way that it doen't require 6 points to be strictly 90 deg apart it has rather big margin. I would rather implement a "Calibrate level" button in Configurator.

oleost, 2016-05-09:
@digitalentity Sounds good.

enrique15241524, 2016-05-11:
Thanks i solved to problem.....

Regards
Murat

closed 2016-05-11 by enrique15241524