iNavFlight/inav #191: Mag is only read at 10hz, why not full 75hz?

theArchLadder, 2016-05-09:
https://github.com/iNavFlight/inav/blob/master/src/main/scheduler/scheduler_tasks.c#L97
Looks like baro runs at 20hz, gps at 25hz even though it only updates at 10hz. But the mag only runs at 10hz, why?

The hmc5883 mag (most popular one) is capable of running at 75hz

It may not be a huge advantage while flying, but i do think this would make a big difference during mag calibration. Each axis have to point to and from magnetic north, and with the low 10hz it seems like the closes reading can be several degrees of from magnetic north, and thus would make a bad calibration.

I guess my main point is that reading the mag at a faster rate would have some benefits, and i can't really see any disadvantage of doing that. Am i missing something?

theArchLadder, 2016-05-09:
Looks like the 10hz is hardcoded in at least these places:
https://github.com/iNavFlight/inav/blob/master/src/main/scheduler/scheduler_tasks.c#L97
https://github.com/iNavFlight/inav/blob/master/src/main/sensors/compass.c#L71
https://github.com/iNavFlight/inav/blob/master/src/main/drivers/compass_hmc5883l.c#L294

digitalentity, 2016-05-09:
Case 1: in flight
Compas is used to slowly correct for gyro drift. No need to get it updated at 75Hz, it should read "approximately north". IMU convergence on compass is very slow - may take several seconds. We might even get away with 1Hz update rate

Case 2: calibration 
Even with 360dps rate (1 rotation per second which is unlikely when calibrating compass) at 10Hz we still get readings 36 deg apart which is more than adquate separation.

The downside of reading the compass faster is that the compass chip is (almost) always located at slow I2C bus and reading it takes roughly 190 microseconds. At 10Hz that's negligible 0.2% of time while at 75Hz we will spend ~14% of time reading data that are not going to improve flight performance. I think this is a good reason to keep compass updates slow.

However this line https://github.com/iNavFlight/inav/blob/master/src/main/sensors/compass.c#L71 is a remnant of pre-scheduler code and can be removed, thanks.

theArchLadder, 2016-05-09:
Well, thanks for explaining so well why i was wrong :)
Did not realize it took so much time to read the sensor.
I also tried to do some calculation on how forgiving the calibration is, this is the code used (processing):

```
void setup() {
  println("error in:");
  println("deg   mag     compass");
  for (int i=0; i<=90; i+=5) { //0 to 45 degrees away from magnetic north
    //max mag value is around 160 in my location
    float milliGaussError = 160 - (160*cos(radians(i)));
    float compassError = degrees(atan2(milliGaussError, 160));
    println(nf(i,2),"  ", nf(milliGaussError,3,1),"  ", nf(compassError,3,2));
  }
}
```

Output:
```
error in:
deg  mag 	 compass
00 	 000.0 	 000.00
05 	 000.6 	 000.22
10 	 002.4 	 000.87
15 	 005.5 	 001.95
20 	 009.6 	 003.45
25 	 015.0 	 005.35
30 	 021.4 	 007.63
35 	 028.9 	 010.25
40 	 037.4 	 013.17
45 	 046.9 	 016.32
50 	 057.2 	 019.66
55 	 068.2 	 023.09
60 	 080.0 	 026.57
65 	 092.4 	 030.00
70 	 105.3 	 033.34
75 	 118.6 	 036.55
80 	 132.2 	 039.57
85 	 146.1 	 042.39
90 	 160.0 	 045.00
```

Looks like you where right again, as long as my calculations was correct, getting the closest reading as far off as 20 degrees away from magnetic north only gives 3.45 degrees error in the compass reading... I have had issues with doing consistent mag calibrations though, but that could be for other reasons i suppose. Or maybe it is that hard to get close enough to magnetic north, who knows...

digitalentity, 2016-05-09:
@theArchLadder 
A few days ago I was reconfiguring my miniquad and typed magnetic declination of 12 deg instead of -12 deg for my area. North error was insane 24 degrees but copter did execute poshold with very low toilet-bowling. When doing RTH is did close in on home location by large spiral but eventually it reached home and landed. 
So I guess INAV is rather forgiving when it comes to compass errors - as long as heading is pointing in somewhat correct direction you should be fine. Accuracy will degrade though.

DzikuVx, 2016-05-09:
I also observed that iNav is forgiving for MAG error. I'm not even aligning mine to the quad axis and it flies perfectly. I'm just pointing it somehow in parallel to front and it works. Error is perhaps up to 5 deg.

digitalentity, 2016-05-09:
This is actually somewhat strange as other firmware - APM, PX4, DJI - is very intolerant to compass errors. Strange, but good :smile:

theArchLadder, 2016-05-09:
I agree with these observations, for example i have had times when i do RTH and it looks like the compass is off by 20-30 degrees, so it starts flying in the wrong directions but then turns towards home and lands without issues, very cool that it's so forgiving!

Putting a error in mag declination sounds like a good way to test how tolerant inav is to compass error, thanks for the advice!

stronnag, 2016-05-09:
@digitalentity , may I commend ```INAV_ENABLE_AUTO_MAG_DECLINATION=1``` to prevent wild declination errors :). In the scheme of things, the approximation error is negligible.

digitalentity, 2016-05-09:
@stronnag, granted. The new #define is called NAV_AUTO_MAG_DECLINATION and it's enabled on SPRACINGF3 and SPARKY targets (the ones that have enough flash to store the lookup table).

digitalentity, 2016-05-12:
I believe we've finished this discussion. @theArchLadder feel free to reopen if you think it's necessary.

closed 2016-05-12 by digitalentity