iNavFlight/inav #192: Board - Naze rev6 support

xalex88, 2016-05-09:
Hello,
sorry in advance for my English. :-)
When supported INAV the NAZE32 rev6?

Best regards

digitalentity, 2016-05-09:
Naze32 rev6 is supported by NAZE target.

xalex88, 2016-05-09:
however, I can find gps fixed nobody. with cleanflight already

digitalentity, 2016-05-09:
Can you explain more, what's the issue about?

xalex88, 2016-05-09:
i didnt have GPS with INAV.
With Clmeanflight i have 7 Sat's
[image]

DzikuVx, 2016-05-09:
@xalex88  iNav does no longer reports each individual sat in Cleanflight Configurator.
Instead, it reports there only HDOP value. Let GPS lay little longer to get the fix and you will notice value 99 drops down as fix quality increases.
Or you can use [iNav Configurator](https://github.com/iNavFlight/inav-configurator) instead

digitalentity, 2016-05-12:
Not an firmware issue. Closing.

closed 2016-05-12 by digitalentity