iNavFlight/inav #196: RTH: Does not keep altitude with stick arming and zero throttle

theArchLadder, 2016-05-10:
As discussed in #157 

I just tried RTH with zero throttle now.
It does not change the heading at all (i guess this is expected in this case).
It goes up in altitude, but it does not care about altitude when flying home so it sort of slowly falls and crashes at the home location.

I'm not sure if i have tried zero throttle for the whole RTH before, so hard to say if this bug was introduced recently or not.

digitalentity, 2016-05-11:
Fixed. Looks like manual altitude control was allowed in RTH mode which resulted in descent during return phase with zero throttle and is apparently not safe.

closed 2016-05-11 by digitalentity

theArchLadder, 2016-05-11:
Just did a test flight and it does hold altitude now. :+1: 
It's a bit wobbly though, i guess this is because i-term is disabled for rate-pid-controller...?
Perhaps airmode would be good to have during RTH?

commit 0592f9f8 (2016-05-11): NAV: Disable manual altitude control in RTH mode. Closes #196