iNavFlight/inav #197: Move from arbitrary rates to degrees per second in FP-PID controller

DzikuVx, 2016-05-10:
Current rates implementation of rates for roll/pitch/yaw uses hard to understand arbitrary numbers like 0,10. How much rotation 0,10 is? It is 300 degrees per second. And how much 0,20 is? Twice 0,10? Nope, it is 400dps. Plus, you can not go below 200dps.

Idea is to drop those arbitrary values and introduce absolute rotation speeds at max stick deflection. This approach is used for example in OpenPilot/LibrePilot and is very easy to understand. You set 360dps for roll/pitch and 90dps for yaw and you know you can do roll in 1s and full yaw rotation in 4 seconds.

This would break CF compability, including Configurator but it is worth it

theArchLadder, 2016-05-10:
I agree! I used OpenPilot/LibrePilot for a short while before, and i miss this feature!

I also think RC Rate should be removed and only use roll/pitch/yaw rates in deg/s, and perhaps expo for each of those.

DzikuVx, 2016-05-10:
@theArchLadder to be honest I do not understand RC rate but I agree that if one rate would be enough.
But I do not agree  on joined roll/pitch rates. I tend to have the different for example
I did some research in the code and discussed few things with @digitalentity and looks like it will be pretty simple to modify PID contrtoller. It internally works in dps already.
Other topic is configurator :)

theArchLadder, 2016-05-10:
AFAIK: RC-rate + Roll-rate = total rate for roll, same for pitch and yaw.

I was unclear. I meant that i would like an option to lock roll/pitch

DzikuVx, 2016-05-10:
Ah... That would have to be configurator issue

DzikuVx, 2016-05-11:
I did some thinking and looks like we can move to dps without totally breaking CF compatibility. Yes, rates would have to be retuned, but original Configurator still would be usable without changes.  
Current rate would not store arbitrary value but dps in tens. So rate 36 (0,36 in configurator) would mean 360dps.

digitalentity, 2016-05-11:
Yes, the same way as with align_board_* values. INAV Configurator will multiply by 10 and display correct dps values but original CF configurator will still be usable.

We also need to make this fool-proof - constrain the rates to sane values like 100-1800 dps. Lower limit of 100dps will keep the quad controllable if user accidentally or deliberately enters zero rate. Upper limit of 1800 dps is just below what gyro can handle (and this is also a hardcoded limitation in PID controller)

DzikuVx, 2016-05-26:
OK, since #204 is ready and waiting to be merged, I'm closing this ticket

closed 2016-05-26 by DzikuVx