iNavFlight/inav #198: Create a way to list possible alternatives for an setting in CLI

digitalentity, 2016-05-11:
Something like this:
```
set mid_rc=?
Arbitrary value in range [1200, 1700]

set rc_smoothing=?
Value from a list: ON, OFF
```

theArchLadder, 2016-05-11:
It would also be more user firendly to get that info when entering a invalid value
```
set mid_rc=2500
Invalid value. Arbitrary value in range [1200, 1700]
```

digitalentity, 2016-05-11:
Agreed.

Linjieqiang, 2016-05-11:
Nice.

stronnag, 2016-05-11:
So say 150 CLI variables potentially affected, some redundancy, so be generous, assume 40 help strings x 30 say bytes per string. 1200 bytes of flash. F3 only material.

oleost, 2016-05-11:
I'm no programmer.

But would it not be possible to add this to the iNav configurator instead of the firmware?

Just have the iNav configurator interwene when entering a command that way?

theArchLadder, 2016-05-11:
All the max/min and list values are already in inav, only two help strings should be needed, one for max/min and one for a list of values, and the just grab max/min/list from settings... At least i assume this is how it will be done, i'm sure @digitalentity will give us the answer tomorrow!

Having this in the configurator would be a nightmare IMHO, because then we would need to update the configurator every time we add, remove or change max/min/list for any setting in inav.

theArchLadder, 2016-05-11:
Oh... I was digging around in the code and it turns out this feature partially already exists!
This command will print a list with all settings, current value, min value and max value:
`set *`
It does not print out the different values for lists like HIGH_G or LOW_G though.

digitalentity, 2016-05-12:
I'm closing this issue, discussion will continue in #203

closed 2016-05-12 by digitalentity