iNavFlight/inav #200: Refactor yaw_control_direction

DzikuVx, 2016-05-11:
Current way of using yaw_control_direction is strange. Instead of being applied on mixer or even PID output level, it is applied on rcCommand and MAG_HOLD controller separately in a different way: in rcCommand it is inverted, in MAG_HOLD it is not inverted.

It would be good idea to unify it, possibly without modifying rcCommand

digitalentity, 2016-05-11:
Yes, it's confusing. I would either apply it to rcCommand[YAW] only. It's a direction of Yaw control via the stick. Positive rate will still be is still clockwise rotation, but we would be able to choose wether "yaw stick to the right" is CW or CCW.

DzikuVx, 2016-05-11:
Since new MAG_CONTROLLER is no longer modifying rcCommand it's either has to be done twice or after PID loop

digitalentity, 2016-05-11:
I think we can simply drop the yaw_control_direction. If you have your
motors or servos the opposite way you can reverse direction via custom
mixer/servo mixer. Thats yaw_control_direction is actually confusing and
not really useful. What do you think? Drop it?

DzikuVx, 2016-05-11:
I love to drop useless features :)

digitalentity, 2016-05-11:
Me too. They free up space for something useful :smile:

DzikuVx, 2016-05-11:
solved by #195 , closing

closed 2016-05-11 by DzikuVx