iNavFlight/inav #202: Revisit filtering defaults

digitalentity, 2016-05-11:
There are alot of complaints about FP-PID being much worse than mwrewrite. After looking into the issue I figured that since 1.0 we significantly changed filtering defaults. FIR filtering in 1.0 was much more restrictive than new BiQuad filtering. We also changed default hardware filtering (gyro_lpf) setting. Seems like older defaults were better suited for bigger quads while also allowing miniquads to be flyable.

We need to revisit filtering to improve default-setting performance on various types of machines.

DzikuVx, 2016-05-11:
gyro_lpf - here I have mixed feelings. While default 42Hz might be safer than 98Hz, it greatly improves acro performance. ON the other hand, noisy frames would suffer a lot. My 600mm flies great on 98Hz but it is not very noisy. For my 250 I had to go back from 188Hz to 98Hz after damaging one of bearings

digitalentity, 2016-05-11:
Changes since 1.0:
Filtering:
gyro_lpf = 42HZ -> gyro_lpf = 98HZ
gyro_soft_lpf = LOW -> gyro_soft_lpf_hz=60 (roughly the same filtering)
acc_soft_lpf = MEDIUM -> acc_soft_lpf_hz = 15
dterm_cut_hz = 30 -> dterm_lpf_hz = 30

PID defaults also changed (if scaled to the same units):
P is slighly lower in 1.1
I is 3.5 times lower in 1.1
D is roughly the same

Two most significant things that changed:
1) FP-PID uses different approach for D-term + BiQuad filter (much more phase shift and therefore much less efficiency). Addressed in #201, should be more tunable when it's merged.
2) Hardware gyro_lpf changed from 42Hz to 98Hz, while this is benificial for miniquad tuning, this is a big load of sh*t for bigger quads with large slow props. I.e. for a 450-sized, 920kv, 9.4x4.3 props, 50% throttle motors spin at ~4000rpm = 67Hz which is well within 98Hz range. On that quad gyro is simply picking rotation from props, not only rotation of the quad.

digitalentity, 2016-05-11:
Suggestion - make 42HZ a safer default for gyro_lpf.

DzikuVx, 2016-05-11:
@digitalentity I did not changed that much, prescaling changed too. If I remember correctly, I is bigger now

digitalentity, 2016-05-11:
@DzikuVx just recalculated to dps. Default kI on Rewrite is ~7.1, on inav it's 2.0 - lower :smile:

DzikuVx, 2016-05-11:
Ah, comparing to Rewrite. In that case it is fine. On the other hand I still prefer LuxFloat over Rewrite :)

digitalentity, 2016-05-11:
I prefered Lux as well, but people seem to miss Rewrite :smiley:

digitalentity, 2016-05-15:
Reverted to 42Hz default LPF in a8a9ba82c54b3e5b144d471373201c2ca9f36300

closed 2016-05-15 by digitalentity