iNavFlight/inav #205: RC Rate - is it really needed?

DzikuVx, 2016-05-13:
While working on #204 I got to a conclusion, that RC Rate is useless. It is used only to build stick curve as a prescaler. With PID rates in dps I see no reason for such a thing. Does anybody knows a good reason to have it? Some radio issues? MultiWii experiment when there were no real rates? Honestly, this setting makes no real sense to me...

theArchLadder, 2016-05-13:
Perhaps the idea was to be able to tune roll/pitch/yaw rates with only one setting and then fine-tune with induvidual settings for roll, pitch and yaw...?

But yes, i agree, this is useless and confusing. Only having separate roll, pitch and yaw rate settings in DPS (degrees per second) would be more user friendly!

skaman82, 2016-05-13:
With roll pitch and yaw rates being zero by default, you can make your quad even a bit slower by lowering the RC rate - good for filming or rookies.

theArchLadder, 2016-05-13:
That may be a feature now, but is no reason for keeping rc rate.
The proposed change should allow roll, pitch and yaw rates to be set to 90dps or something like that if one wants super slow rates.

DzikuVx, 2016-05-13:
Exactly. With absolte rate 60dps for example there is no need for rc rate

stronnag, 2016-05-13:
RC Rate is a hang over from the original MW PID controller. All the modern controllers just consider RATE and axis rates as additive, the original MW controller treated them differently. Thus there is no need for it any more.

DzikuVx, 2016-05-13:
So it will be gone....

DzikuVx, 2016-05-13:
And it is gone in f112a127820148690cc782f17b604a19def0be3a

digitalentity, 2016-05-13:
@DzikuVx @theArchLadder @stronnag 
If we would make very low roll/pitch/yaw rates possible (like 60-90dps) `rcRate` really becomes obsolete. I can't remember ever changing `rcRate` :smile: 
Thanks for addressing this. I wonder how much more useless/obsolete code is there?

DzikuVx, 2016-05-14:
A lot. For example look at https://github.com/iNavFlight/inav/blob/master/src/main/mw.c#L135 Duplicated code. I will also get rid of those in #204 

Only getting rid of rc_rates freed ~ 300 bytes of flash

closed 2016-05-14 by DzikuVx