iNavFlight/inav #207: GPS: Better protection from GPS glitches

digitalentity, 2016-05-13:
We need to devise protection from GPS jumps (aka glitches) similar to one described in  #206.

bk79, 2016-05-14:
I would use a smoother...if (actual pos - previous pos)>5m wait for 10 sec
to have it confirmed otherwise skip the measure

csurf, 2016-07-20:
- calculate the rate of change / position delta from last known good position to current reading
- Provide a user-configurable tolerance parameter that sets the 'maximum allowed change in GPS position' in meters per second or something 
- If the delta from the last known good position to the current GPS position is above the threshold, mark it as 'bad' / discard it, and stick with the last known 'good' position value. 
- Lower values for this parameter would prevent big position jumps due to glitches, but also wouldn't allow for high-speed vehicles (would be good for slow AP quads, bad for fast planes). Higher values would allow for higher speeds but more glitchiness to get through, which is probably ok for planes.
- perhaps find a way to integrate a 'backup metric,' such as HDOP or sat count, to use when evaluating the position data? Ex., if sat count increases & HDOP improves, perhaps a newer, more precise position reading might appear to be a 'glitch' when compared to the older, less precise data if there's a large difference between the two.

thoughts? I was just brainstorming here...

kronosol, 2016-07-23:
Can't we just compare the acceleration from ACC with the change of GPS position?
So we check the GPS Update with the ACC data.
For example, the GPS glitches 5m in 1 sec, which mean a acceleration of 5m/s². But the ACC only measures a acceleration of 0m/s². In that case inav should disable PosHold and climb at least 5m, so the quad can't drift into someone standing near to it.

kronosol, 2016-07-23:
Or in case of a nearly relase of Inav FC, how about a second GPS module? The GPS with the position which is nearly to the last 5 average positions are true.

digitalentity, 2016-07-24:
I have some ideas for better glitch detection, but this will have to wait until after 1.2 release.

closed 2018-05-27 by stale[bot]