iNavFlight/inav #208: [Flashing] Prevent damage of servos when updating to new version.

sppnk, 2016-05-15:
When updating FC, the default configuration reverts to multicopter. If there are servos attached to the FC the high frequency outputs can damage them before being able to configure all the setup as airplane. 

Also, depending on the setup and power supply, as the current is too much, the FC is not able to restart and comunicate through serial USARTs to be configured.

In this case, the only way to reconfigure for a new version is disconnect all servos and flash as normal. In many airplanes you have to unasemble everything to do this.

Should be good not to output any signal until everything is configured. Sort of a switch in configurator or something like that.

oleost, 2016-05-15:
Why not simply flash without main flight power connected?

Never had an issue.

sppnk, 2016-05-15:
I am not able to do that. If I try to flash only through FTDI, the 5V go to the servos, so the FTDI adapter or the USB port of PC cannot supply that. May be this is only a problem in some FCs.

digitalentity, 2016-05-15:
Yes, this might be a good safety feature.

sppnk, 2016-05-16:
May be a quick solution to set after-flash default to airplane, not multirotor :-)

oleost, 2016-05-16:
Ah good not. Dont think multirotors would agree with you there.

jarno83, 2016-05-16:
I think that would be good if after flashing you'll get a window of main options like multi or plane, fc orientation or so a basic calibration and so on :)

sppnk, 2016-05-16:
@oleost , sometimes I think we should have a diferent branch, ha ha. :-)

oleost, 2016-05-16:
But to keep it simple, I think my idea works fine.

No signal out until it detects that adanced accelerometer calibration is done.

sppnk, 2016-05-16:
Yes, that should do. This way everybody should have to calibrate well their aircrafts. For me should be ok.

sppnk, 2016-06-28:
For now I am building my own versions just changing MIXER_QUADX for MIXER_AIRPLANE in config.c file, but a better solution should be found. Otherwise is very annoying to unasemble the planes for flash the FCs.

Other thing that should be take in consideration, is that when flashing, the GPS has to be disconnected, as it is not possible to connect again throught USART until you disconnect it. Once GPS is disconnected I can configure everything, including GPS (usually by CLI dump, but in a new model it will be in another way). This is less annoying, as GPS is more accessible, but leads to problems to newcomers. It happens to me in CC3D (naze dont remember).

May be a 'just flashed right now" flag should do for both issues (outputs and gps).

oleost, 2016-07-17:
As an easy fix, would it be possible to just create a new Empty mixer, which only outputs throttle on output 0, (Or even nothing at all if possible) And then just make it the new default mixer?

digitalentity, 2016-07-17:
Suggestion: implement a "MOTOR_CONTROL_ENABLED" feature. It will be always disabled when config is erased and it will prevent all PWM outputs to be configured.

We have a free bit in `features` (previously occupied by FEATURE_INFLIGHT_ACC_CAL) - we can re-use it.

sppnk, 2016-07-18:
@digitalentity . I like that solution.

oleost, 2016-07-18:
Probably a god solution.  I just proposed something I thought would be an easy fix,  start with no mixer as default and then have people choose the correct mixer when configuring.

sppnk, 2016-07-18:
And I like much more the @oleost solution, so we could still have free the  bit in features for future use.

digitalentity, 2016-07-19:
The problem is mostly with refresh rate. Servos are designed to operate at 50Hz, while brushless motors on quads run on 490Hz in PWM mode. If you start with no mixer, you still have idle PWM signal on motor outputs that will damage your servos.
I think that delay PWM output initialisation (no signal until configured) is the only bullet-proof solution to protect servos from damage.
And we have a whole lot of free bits in `feature` actually - its a 32-bit number.

oleost, 2016-07-19:
Okey,  sounds good.

sppnk, 2016-07-19:
Ok then. Please add this feature to next 1.2 release. Thanks.

digitalentity, 2016-08-18:
This is closely realted to #397 and will be done along with it first thing after 1.2

oleost, 2016-10-17:
@digitalentity See #397 is now finished and merged.

Is this possible to implement now? (Not to 1.3 but to development)  I have two planes where I have to build my own firmware with airplane as default mixer due to servos are powered by USB on spracingf3mini and by the FTID adapter on my flip32.

digitalentity, 2016-10-17:
@oleost yes, with #397 merged this can be addressed as well. As I have a few airplanes as well this will come in handy.

sppnk, 2016-10-17:
Lately I am not testing firmware (I have been involved in having a working FPV system), but when I did during last months I also had to build my own versions in order not to unasemble the planes. Otherwise I had servos burnt.

Good to hear you are working in this.

digitalentity, 2016-10-17:
@oleost, @sppnk - long awaited feature is now available for you to test and use.

sppnk, 2016-10-17:
Great. So if I want test this I have to compile develop branch, not master, isnt it?

digitalentity, 2016-10-17:
@sppnk you should check out the `pwm-output-enable` branch.

sppnk, 2016-10-17:
oh, sorry I didnt realized of that. Thks.

oleost, 2016-10-17:
@digitalentity Bench tested on my airplane and did the trick for me, spracingf3mini target.

@sppnk close this and we conuintue conversation over test results at pull request?  #706

digitalentity, 2016-10-18:
PR to implement this - #706. Closing

PR #706 (2016-11-07): PWM output disable/enable feature
When updating FC, the default configuration reverts to multicopter. If there are servos attached to the FC the high frequency outputs can damage them before being able to configure all the setup as airplane.

This PR adds a new feature `PWM_OUTPUT_ENABLE` that is a flag to actually allow anything to be output to motors and servos. If the feature is disabled all outputs are pulled low.

Feature `PWM_OUTPUT_ENABLE` is disabled by default so nothing will fly or move before user decides so.

At the moment this feature can only be enabled via CLI, configurator supp [... cut]
closed 2016-10-18 by digitalentity