iNavFlight/inav #209: BUG iNav V1.1

raoul596, 2016-05-15:
I have flashed the naze.hex on my flip32+. In the previous version of iNav 1.0.1 it worked fine. Now I updated the firmware to iNav 1.1, I can't select the hex + frame type, it always goes back to the quad X configuration. I tried to set it in the CLI and I got the same bug.

What am I doing wrong?

oleost, 2016-05-15:
#143 is it regular hex6 or something else your trying to select?

They have removed alot of predefined mixes to save space, if it's one of those your trying to select you need to use custom mix instead.

https://github.com/iNavFlight/inav/wiki/Custom-mixes-for-exotic-setups

digitalentity, 2016-05-15:
@raoul596 
Hex+ mixer is considered "uncommon" and was removed in #143 for all F1 targets (NAZE, CC3D). You can still configure Hex+ mixer by using cmix:
```
mixer CUSTOM
mmix reset
mmix 0 1.0 -0.866025 0.5 1.0        # REAR_R
mmix 1 1.0 -0.866025 -0.5 -1.0      # FRONT_R
mmix 2 1.0 0.866025 0.5 1.0         # REAR_L
mmix 3 1.0 0.866025 -0.5 -1.0       # FRONT_L
mmix 4 1.0 0.0 -1.0 1.0             # FRONT
mmix 5 1.0 0.0 1.0 -1.0             # REAR
```

closed 2016-05-15 by digitalentity