iNavFlight/inav #210: Limit yaw acceleration in YAW axis

DzikuVx, 2016-05-15:
in a09f07caf2951ece6a7ebdc9617a202f077c10d8 filterApplyPt1WithRateLimit was introduced. 
Extend MAG_HOLD controller to use it. Also, define default for MAG_HOLD.

Idea: does it makes sense to replace yaw_p_limit and yaw_jump_prevention_limit with limited acceleration on YAW axis? For sure this would make tuning easier

digitalentity, 2016-05-15:
I like your idea of general yaw acceleration limiting - it may also work to complement mag_hold.

However I don't think it will replace `yaw_p_limit` and `yaw_jump_prevention_limit` as they serve a slightly different purpose - they limit the PID response which is generated not only by stick command, but also by external interference - like rotation induced by clipping the gate.

DzikuVx, 2016-05-15:
I thought for a second about it and looks like they can not be replaced after all. You are right here. Acceleration limiting on user input can be added, but will not replace `yaw_p_limit` and `yaw_jump_prevention_limit`

digitalentity, 2016-05-15:
I'm already adding general yaw accel limiting to a separate branch. Stay tuned.

digitalentity, 2016-05-15:
Please see b88a828 and yaw-accel-limiting branch. Code is just a proof of concept. If it works we can remove LPF from MAG_HOLD controller and add 2 more settings - `yaw_rate_lpf_hz` and `yaw_rate_accel_limit`.

DzikuVx, 2016-05-15:
There will only an issue of tuning this thing. Damn... we are changing quite a lot lately :)

digitalentity, 2016-05-15:
Defaults should be set to quite high values - currently there is neigher LPF nor rate-limiting. One wouldn't have to change anything unless his quad looses stability when doing hard yaws, otherwise - reduce `yaw_rate_accel_limit` and sacrifice speed of response until there's no loss of stability. I think we don't even need `yaw_rate_lpf_hz`.

digitalentity, 2016-05-15:
Yes, we change quite a lot. And I must say I like it!

theArchLadder, 2016-05-15:
@digitalentity 
In what way is a LPF helpful compared to only acc-limiting?

DzikuVx, 2016-05-15:
Lpf would make transition smooth while below acceleration limit. 

Previous mixer was not prioritizing yaw. Mixer from Betaflight is.

digitalentity, 2016-05-15:
The new mixer doesn't prioritize anything, it does what it can to satisfy PID command. Prioritizing one axis over another actually is a bad thing.

If we are over the limit of the setup we will end up with I-term winding up like crazy, this is why limited yaw performance is best dealt with in PID controller - the most sophisticated part of flight stabilisation.

DzikuVx, 2016-05-16:
There is smth with new mixer and stability with yaw fast acc/speed. My 600mm was more stable in those conditions on old mixer. So maybe not priority per se, but there is smth in place

digitalentity, 2016-05-16:
New mixer is allowed to adjust throttle (increase or reduce) to satisfy the most of PID output. The old mixer wasn't allowed to do it and PID output might get clipped by mixer.

theArchLadder, 2016-05-16:
@DzikuVx @digitalentity 
What is the difference between yaw_jump_prevention_limit and yaw_p_limit? Both of them seem to limit yaw, so what is the difference in flight behavior between them?

DzikuVx, 2016-05-16:
* yaw_p_limit is PID loop only. It's an absolute limit of error * P gain. 
* yaw_jump_prevention_limit is mixer located. It limits total YAW axis PID output comparing to current stick position. So, on stick at 0 (middle) and yaw_jump_prevention_limit=200 only values between -200 and  200 will be written to motors (well, mixed not written). At stick at 100, -100 and 300 and so on

Lately I increased yaw_jump_prevention_limit on my 600mm to 400 and it improved yaw performance greatly

theArchLadder, 2016-05-16:
Hmm... In the end it seems like both of them limits the difference in motor speed for yaw...
Would you care to give an example of a scenario where they would do different things?

theArchLadder, 2016-05-17:
Rain finally stopped!
Did a test flight to find out the difference between yaw_jump_prevention_limit and yaw_p_limit.
I have to say it's really hard to tell a difference for me, p_limit reduces does sudden changes in motor speed and makes yaw smoother, but so does jump_prevention as far as i can tell, the effect of it was not as big though.

Perhaps my quad has to great yaw capability to really need or notice any difference with the jump_prevention? It's still unclear to me what jump_prevention can help with that p_limit cannot.

DzikuVx, 2016-05-17:
They overlap each other, true.

theArchLadder, 2016-05-17:
It does sound like only having acc limit and rate limit would be more user friendly and easier to tune...

If i yaw very fast back and forth, my quad gains lots of altitude, no matter what i set jump_prevention and p_limit to, i guess this has something to do with my low hover throttle around 1250, so the mixer raises the throttle.

digitalentity, 2016-06-22:
@DzikuVx is this issue still valid? Should we implement Yaw accel limiting?

DzikuVx, 2016-06-22:
@digitalentity I'm not sure. Perhaps, but I do not think it's required not. Latest changes to rates and MAG_HOLD controller improved things so much that even my overweighted 600m quad behaves ok in yaw. I would leave the ticket, only remove 1.2 milestone

digitalentity, 2016-06-22:
Agreed and done.

DzikuVx, 2016-09-15:
With `rate_accel_limit_yaw` this issue is obsolete

closed 2016-09-15 by DzikuVx