iNavFlight/inav #212: Shouldn't mode activation conditions be ANDed instead of ORed?

iforce2d, 2016-05-16:
The configurator lets you add activation ranges on multiple AUX channels for a mode. I had always assumed these conditions would be ANDed to allow many different mode states and make the best use of our limited number of transmitter switches. But when I tried this for the first time, I found out these conditions are ORed. This seems to be the same in original Cleanflight.

I can only assume this is not the intention because it's kinda pointless isn't it? Why would I want to use two different switches to activate the same mode... I might be missing something here, but fwiw I tested this change I made which will AND the conditions and it seems to work ok, at least for this case I was using:

AUX1 low = full manual
AUX1 mid = POSHOLD
AUX1 high + AUX2 low = WP
AUX1 high + AUX2 high = RTH

http://pastebin.com/XUbf7LNR

digitalentity, 2016-05-16:
Good find, thanks!
I'll check and fix.

theArchLadder, 2016-05-16:
I agree with this! I just want to share my solution for "to many flight modes and to few channels".

This is how i have solved it for now with my 6 channel radio:
AUX1 low = rate mode
AUX1 mid = poshold+althold
AUX1 high = poshold+althold+wp_nav+gcs_nav

AUX2 high = RTH

wp_nav does not activate if there is no wp-mission, and afaik it doesn't really hurt to have gcs_nav active while doing wp-missions. This way i can have both wp and gcs_nav (follow-me etc) on the same switch position! It would be nice to also get angle mode in there somehow though. And if i want to do follow-me after a wp-mission, I'd have to power cycle.

DzikuVx, 2016-05-16:
I would not call it a bug, and the way I see it, it's working fine. I have my 5 "flight setups" driven by 2 switches mixed to 1 radio channel. That means, that for example ALTHOLD is activated in more than 1 range of AUX1. 

How I see it, it might make sense to give a user ability to choose AND or OR. Or at least AND in different channels, OR in 1 channel.

oleost, 2016-05-16:
I would not call it a bug either. Working as intended.

But I agree with  @DzikuVx that it could be user selectable. But if this is something iNav should tackle or Cleanflight then merged with iNav is another thing.

theArchLadder, 2016-05-16:
Agreed
For example, lets say i have poshold on AUX1 and WP on AUX2, now if i forgot to upload the wp-mission or something similar, i might want to get poshold when WP fails, so i would set poshold to AUX1 high OR AUX2 high.

@iforce2d Can you show an example where AND would be useful? Your example does not make sense to me, use WP on "AUX1 high" and RTH on "AUX2 high", RTH will overrule WP.

theArchLadder, 2016-05-16:
I think i just came up with an answer to my own question, this is how AND could be useful

AUX2 low AND:
AUX1 low: rate mode
AUX1 mid: air mode
AUX1 high: angle mode

AUX2 high AND:
AUX1 low: poshold
AUX1 mid: wp/gcs nav
AUX1 high: RTH

So the AND function would allow 6 different flight modes instead of just the 4 shown in my first example, the only issue i see with this is that rth/poshold as a "oh crap"-switch would now require two switches to be changed.

This can get complicated, for instance, if i want to replace angle mode with poshold, i would have to use both OR and AND functions. Not sure how to do a user friendly GUI for this...?

digitalentity, 2016-05-16:
I think we should have a CLI option to choose between OR and AND behavior with default to OR (compatibility). What do you think?

theArchLadder, 2016-05-16:
@digitalentity Not being able to use both OR and AND at the same time will not give full freedom, but that would be complicated and require big changes to both inav and the configurator, so i think your idea sounds good enough and would work for most use-cases! Go for it :+1:

digitalentity, 2016-05-16:
Looks like it will require some serious refactoring. Current code simply stops at first matching range. Still, scheduling for 1.2 :smile:

iforce2d, 2016-05-16:
Well I'm not calling it a bug either, I'm mainly just asking if this really is the intended behavior, because it doesn't make much sense.

@theArchLadder Your example using modes which can co-exist without problems is not really a good one, it's only when you want to use modes that don't play so nicely together that separating them via switch is important.

@DzikuVx This issue is about multiple AUX channels so I don't see how that case is relevant. If the transmitter can mix switches into one channel that's great, but not everybody's TX can do that, so this is why (I had assumed) Cleanflight had the option to AND multiple channels in the configurator.

@theArchLadder I gave an example in my first post of why having AND between channels would be useful. Perhaps my example was not so impressive with only four distinct states, but that was the case I have physically verified so I didn't want to claim something I had not checked. For the case I gave, since I only wanted four modes and RTH overrides WP I suppose I could have set it up this way (with OR between channels) to achieve the same result:
AUX1 low = full manual
AUX1 mid = POSHOLD
AUX1 high = WP

AUX2 low = nothing
AUX2 high = RTH

But this relies on prior knowledge that RTH will override WP which is not very user friendly (I did not know myself until reading this thread), so it's a special case not applicable to all modes.

Okay, so it's understood this is not a bug (original Cleanflight has been like this for a long time without any fuss), but my question from the first post still remains: why would I want to use two different switches to activate the same mode? For example if I can use AUX1 to enter RTH mode, what good does it do me to add an OR condition with AUX2 to enter RTH mode, so that RTH can be entered with either AUX1 OR AUX2? It just seems like a waste of switches.

I think the bottom line is if my TX is capable of sending 6 distinct states, the flight controller should be capable of running in 6 distinct states, without relying on modes to override each other or be compatible in special cases (eg. when no waypoints are set).

@digitalentity Yes, I think the ideal resolution is to have this as an option in the configurator, with current behavior as default. It would be perfect if the AND/OR logic was selectable per mode, but that seems like quite a bit of work, when having a singe option across all modes would be good enough for almost all cases.

commit 394a2882 (2016-07-25): Allow ANDing of multiple channel ranges in mode activation conditions (#364)

* Allow ANDing of multiple channel ranges in mode activation conditions (CLI: set and_mode_conditions=ON)
* Moved mode logic operator to Profile (modeActivationConditions are part of profile already); Created a lookup table for logic operator
* Closes #212
closed 2016-07-25 by digitalentity