iNavFlight/inav #213: Switch profiles based on battery cell count

eziosoft, 2016-05-16:
As I fly with two sets of batteries, 3S and 4S I would love to see automatic profile selection based on cell count (battery voltage)
It will allow to tune the quad for each battery without thinking of changing a profile.
What do you think.

DzikuVx, 2016-05-16:
I'm still working on #74 that will expose battery level as virtual AUX. Not the same, but if cell count also would be exposed as virtual AUX level and profile switching made mode selectable, that would might work too and give possibilities to switch profiles with switches, not only sticks

closed 2018-05-27 by stale[bot]

kn8alpha, 2019-01-25:
Can this be done by using setConfigProfile() from any function, for example some function that detects cell count, and apply the settings when disarmed?