iNavFlight/inav #216: Optical flow support

digitalentity, 2016-05-17:
I'm thinking about supporting a cheap OpFlow ADNS3080 mouse sensor, but it should be connected to SPI bus. We have a few options:

1. Create drivers for SC18IS602B chip - it's a I2C-SPI bridge and with its help we can connect opflow to I2C bus like other sensors. Downside - bandwidth. I2C is usually rather busy with accel/gyro already.
1. Create a Arduino-based board to read data from OpFlow sensor and present it to the FC via Serial port. We can use a special MSP message for this. Downside - we need an extra serial port to use i.e. GPS and OpFlow together.

Ideas?

Linjieqiang, 2016-05-17:
Great~!I will keep learning your code.

DzikuVx, 2016-05-17:
Hmm... can we make our own FC board with 5 UARTs please?
But seriously, I'v been thinking about similar problem and the best solution I've found is MSP hub/bridge/proxy with USB port.
It allows to connect multiple MSP devices, with different speeds and passes messages to FC and then passes answer to device that asked for info. If messages would be queued then response can be presented to proper external device. But building smth like this is as probable as FC with 5 UARTs

DzikuVx, 2016-05-17:
And from sensor docs: "Distance from lens reference plane to surface: 2.4mm (typical)". So what would be this thing useful for?

digitalentity, 2016-05-17:
If equipped with a lens it can be used up to several meters distance. http://www.ebay.com/itm/-/141282083588

DzikuVx, 2016-05-17:
And now we are talking! How do you think, what kinds of refresh rates would be required here? 10Hz?

theArchLadder, 2016-05-17:
Searching youtube for "optical flow sensor" gave some interesting videos.
https://www.youtube.com/watch?v=bbS1N--bT0w
https://www.youtube.com/watch?v=Ox8oI7nzSPw

Hard to tell but it does not look more accurate than a ublox M8N gps, but i guess that could be up to bad tuning. At least it allows indoor poshold...

digitalentity, 2016-05-17:
@DzikuVx the more the better - we don't have an absolute position measurement with this sensor - only relative motion (velocity) in arbitrary units (if we have sonar sensor - in proper cm/s). I think at least 10Hz is absolutely required. 50Hz should give better results.

@theArchLadder 
That's likely optical flow without any assistance. We will benefit from inertial position esitmation and quite possibly have much better results. One caveat - we need working sonar to correctly process optical flow data. On miniquads sonar is almost useless - noise from propellers overloads it at altitudes >50-70 cm.

digitalentity, 2016-05-17:
Ok, got ADNS3080 working with Arduino. It really needs a good light, but it works. A first step to have optical flow in INAV.

digitalentity, 2016-05-17:
Actually, in well-lit environments it's rather stable. I moved the sensor over my desk a few dozen times and accumulated the flow values - error is minimal. Now a big task of attaching it to test copter and make it fly!

theArchLadder, 2016-05-17:
I'm guessing it requires some clever code to compensate for when the copter tilts over and it look like the ground is moving, but the copter actually just changes attitude...?

digitalentity, 2016-05-17:
Yes, it certainly does. 
The most challenging part is making it work with our position estimator and provide readings in meters. Also ADNS3080 will likely return invalid data when copter is yawing.

Linjieqiang, 2016-05-17:
So nice~[image]

martinbudden, 2016-05-20:
This looks very interesting.

The new sonar code allows easy implementation of other rangefinders, for example the [TeraRanger One](http://www.teraranger.com/products/teraranger-one/) infrared rangefinder or the PulsedLight LIDAR-Lite rangefinder (although these may no longer be available now that Garmin has bought PulsedLight). These rangefinders avoid the problems sonar has with propeller noise.

Anyway, it's on my todo list to implement drivers for these rangefinders, but in the medium rather than short term future (I have to get hold of one first).

johnsilvester, 2016-10-19:
How's the development on this going? Any luck with the ADNS3080 Optical Flow Sensor?
Excited to hear!

Jivefunk, 2016-11-04:
Just created an account to follow this thread. I also need this implementation for my quad.

digitalentity, 2016-11-04:
This is a very slo-mo task, sorry. Eventually we'll see opflow support, but probably not very soon.

kitchung, 2016-11-04:
I'd love to see this implemented too!

digitalentity, 2016-11-04:
Work is being slowly done here https://github.com/iNavFlight/inav/pull/434

kitchung, 2016-11-04:
@digitalentity Thanks for the info, subscribing there now.

bk79, 2017-05-05:
@digitalentity playing with a cheap defective mouse from banggood I've found that inside it was using an a flow sensor marked KA8
http://www.pixart.com.tw/upload/PAW3204DB_SPEC_V40_20091207_20100319090638.pdf
looks like it's uses a serial protocol probably is worth trying using I2C protocol to read it

digitalentity, 2017-05-09:
@bk79 protocol of that sensor seems to be half-duplex SPI and it's possible to connect it to a dedicated SPI interface. Worth trying.

bk79, 2017-05-11:
@digitalentity  it's a two wire communication, spi half duplex shouldn't be at least with three?

10ishq, 2017-11-17:
@digitalentity , a separate board with open source hardware with 3080 optical flow sensor which could calculate horizontal movements as well as compensates for tilt using a gyroscope+magnetometer+accelerometer, with a separate processor, cortex M0 seems a good idea, just like ardupilot folks did with PX4Flow but cheaper. your thoughts?

markandkymward, 2019-03-11:
Anyone interested in PX4FLOW sensor?

vc, 2019-03-11:
very important for me

0120shrey, 2019-04-28:
Hi! I'm here after reading about the experimental OPFLOW mode in the 2.0.0 release notes. After reading through a lot of threads on github, I now know that it needs an ADNS3080 sensor, and a sonar (I don't know which). But I couldn't find anything on how to actually connect and use it. I have a quad running on Omnibus F4 clone, which i'm willing to crash! Any support would be great!

georgekucher, 2019-04-28:
+1 I have matek f722 and us100 sonar and desire to buy opt flow sensor. Which one?

closed 2019-06-27 by digitalentity

whoim2, 2019-08-31:
Тоже хочется поэкспериментировать с простыми и недорогими датчиками оптическими. Лазерный дальномер скоро приедет, буду пробовать использовать по i2c.
Оптический датчик хотелось бы подключить по spi либо тоже i2c, поскольку все uart заняты.
Готов погрузится в компиляцию прошивки под свои omnibusf4prov3.
Если могу чем то помочь - обращайтесь.

whoim2, 2020-08-27:
Where i may read about msp2 messages, opflow, sonar data?

Alechaicuse, 2023-01-04:
Hello world! I want to connect my OpenMV with mavlink_opticalflow firmware. How can I connect this system to INAV 6 FC? I'll try plugging it in by selecting the options I've already added (PMW3901/CXOF/MSP/FAKE) but I'm not sure it should work...