iNavFlight/inav #217: Missed OpenPilot Revolution support

night-ghost, 2016-05-17:
There is a wonderful small controller Revo Mini, which supported by LibrePilot, recently even Ardupilot (https://github.com/ArduPilot/ardupilot/pull/3874) - but does not iNav.  There are Stm32f4 forks have been made, so why is it not in the list of supported cards?
In addition, this support will allow to compare the behavior of the same copter with different software

digitalentity, 2016-05-17:
Blocked by #130

night-ghost, 2016-06-03:
there is full REVO support - https://github.com/rs2k/raceflight

digitalentity, 2016-06-03:
@night-ghost Who's going to port the code to INAV?

martinbudden, 2016-06-08:
@digitalentity , I've been following the PR to add F4 support to betaflight, see PR https://github.com/borisbstyle/betaflight/pull/452 .

I think if @ledvinap 's PR https://github.com/cleanflight/cleanflight/pull/1421 "IO and EXTI driver" is merged into iNav the the F4 port is relatively straightforward - I'd certainly be willing to have a go at it.

digitalentity, 2016-06-09:
@martinbudden you have a go :smile: 
I won't be able to crack the code for abput 2 more weeks so any help would be very much appreciated!

martinbudden, 2016-06-09:
@ledvinap , would you have time to create an iNav PR for  https://github.com/cleanflight/cleanflight/pull/1421 "IO and EXTI driver" ?

One that is merged in I can start looking at doing the F4 work for iNav.

closed 2016-08-18 by martinbudden