iNavFlight/inav #220: Altitude Hold default throttle behavior.

oleost, 2016-05-18:
Instead of using default 1500us (nav_mc_hover_thr = 1500)

Use throttle value when activating altitude hold, but restrained to between 1200-1650us. (Should be within most ower powered and under powered flying machines.)

Reason: I must guess that most people when activating altitude hold is already at steady hover in most cases.

theArchLadder, 2016-05-18:
I agree that midthr is a bad default, it's very awkward on my 250 quad with hover throttle around 1250.

IMHO it would be even better "altitude hold throttle" would equal nav_mc_hover_thr. This way the throttle behavior would be consistent in althold, even when arming and taking off in althold mode. Thoughts?

oleost, 2016-05-18:
This could work like this this:

1. Activate alt hold.
2. Throttle is changed depending on settings. (Stay like this Issue propose) or jump to nav_mc_hover_thr
3. Netraul point for althold is set based on nav_mc_netraul_althold.

And have only these two settings:

* nav_mc_hover_thr. (Default 0, which deactivates this and just uses current value when activating althold)
* nav_mc_netraul_althold. (Values: midstick, current_stick and equal_nav_mc_hover_thr)

So we could remove "nav_use_midrc_for_althold" "nav_use_midthr_for_althold"

digitalentity, 2016-05-18:
The "nav_use_midthr_for_althold=1" doesn't mean that midrc (50%) throttle would be treated as neutral position. Neutral position is calculated using **thr_mid** setting (whilch is also middle for throttle expo curve). If your quad hovers at 30% throttle you should do the following:
```
set thr_mid=30
set nav_mc_hover_thr=1300
```

However what @oleost is suggesting looks like a viable option. Replace nav_use_midthr_for_althold with enum: MIDTHR (mid_thr value), CURRENT (throttle at activation), HOVER (= nav_mc_hover_thr). This will give more flexibility.

Also, we need to better define behavior when CURRENT is selected but althold is activated at the moment of arming (zero throttle). Apparetly we can't set zero throttle as neutral position - this will result in inability to descend.

oleost, 2016-05-18:
@digitalentity Tought you already adressed the issue when activating alt hold at low throttle combined with using "nav_use_midthr_for_althold = off" ?

digitalentity, 2016-05-18:
Yes, current behavior is use **mid_thr** regardless of settings if throttle is at zero.

theArchLadder, 2016-05-18:
Do you guys see any problem with using nav_mc_hover_thr as neutral position?
nav_mc_hover_thr has to be tuned to get althold to work well anyway. thr_mid does not have to be tuned (i have never touched it).

The "throttle at activation" does not give consistent stick behavior...

oleost, 2016-05-18:
@theArchLadder it doesnt need to be 100% consitent, if you want that the easiest would be spring loaded to 50% throttle.

When most people active GPS assisted mode/althold, they already hovering fairly still.

Weather your neutral is being set to 33% or 38% stick you wont notice much difference. I think its important that when you  active, it stay still, and this seems to confuse many if it does not.

Anyway, with new proposed way, you can get it like you want.

theArchLadder, 2016-05-18:
Well, it doesn't need to be 100% consistent, but it would be nice if it was :smile: 

"When you activate, it stay still", we have CURRENT (throttle at activation) for this, fine with me.

My question is about MIDTHR (mid_thr value) vs HOVER (= nav_mc_hover_thr)
It doesn't make sense to me to be required to tune two settings when only one, that has to be tuned anyway, can be used. What would the benefit of MIDTHR be over HOVER?

digitalentity, 2016-05-18:
MIDTHR is required if my miniquad hovers at 1250 throttle but I want altitude hold neutral point to be at 50% throttle (i.e. spring-loaded stick). You need to understand that pilot's input and controller output to motors aren't directly related. We need to cover as much options as possible.

theArchLadder, 2016-05-18:
Hmm, my impression was that the thr_mid and thr_expo was just used to make the throttle stick less sensitive around hover, not change the stick position of hover. Thanks for clarifying!

I think i need to do some test flights to really understand how thr_mid and thr_expo works.

I do like your idea about one setting with MIDTHR, CURRENT, HOVER! What should be the default?

digitalentity, 2016-05-18:
@theArchLadder I think HOVER should be default - it's equal to MIDTHR if you didn't tune anything, and if you set your hover_throttle right you have smoother transitions to/from althold.

theArchLadder, 2016-05-18:
Just did a quick flight test, 25% throttle stick is hover for me. I don't think thr_mid is intended for changing the hover stick position, let me explain:

1. This made no difference, neither did thr_mid=0.75
2. This made throttle stick less sensitive around hover so it was easier to manually keep altitude, but i still had 25% stick to hover. This seems like the intended use.
3. Note: thr_mid=0. This made my hover stick around 60% or so, but this does not seem like the intended behavior and there is no way to say "i want to hover at 50% stick".

[image]
[image]
[image]

digitalentity, 2016-05-18:
@theArchLadder 
1) will only affect AltHold mode WITH nav_use_midthr_for_althold=1 Set thr_mid=75 and INAV will assume neutral throttle position at 75% throttle.

theArchLadder, 2016-05-18:
That was a answer to 

> MIDTHR is required if my miniquad hovers at 1250 throttle but I want altitude hold neutral point to be at 50% throttle

To me it seems like if i want to use the thr_mid and thr_expo as intended for manual modes, i should set it to 0.25 (hover stick). This will not be compatible if i want to have neutral point for althold at 50%. And if i want to use neutral point=manual hover stick, why not just use HOVER (= nav_mc_hover_thr)? I doubt many people use this thr_mid thr_expo feature? But we all have to tune nav_mc_hover_thr

theArchLadder, 2016-05-18:
This is getting confusing, let me try to clarify, this assumes a quad with 25% hover stick.

**First up, manual mode:**
In manual mode my powerful quad makes my throttle stick very sensitive, so it's hard to keep altitude. I can set thr_mid to 0.25 and thr_expo to 0.5 for example, so that the throttle stick becomes less sensitive, but i will still have a hover stick of 25%. It is not required to tune mid_thr if i'm happy with my stick sensitivity though.

**Okey, lets go o to behavior in althold now:**

- If i want my neutral stick=manual hover stick, it would make sense to use nav_mc_hover_thr because i have to tune it anyway.

- If i want my neutral stick to be at 50%, i would have to change thr_mid, but this would break the functionality of having a less sensitive throttle stick in manual mode.

digitalentity, 2016-05-18:
@theArchLadder ok, now I see the issue. So we should have following options:
MID = stick middle (mid_rc)
CURRENT = throttle at activation
HOVER = nav_mc_hover_thr

digitalentity, 2016-05-18:
And we let the mid_thr and thr_expo alone.

theArchLadder, 2016-05-18:
Good, it was a bit challenging to explain that confusing stuff!

Do you know what is mid_rc used for?
From cleanflight cli docs:

> This is an important number to set in order to avoid trimming receiver/transmitter. Most standard receivers will have this at 1500, however Futaba transmitters will need this set to 1520. A way to find out if this needs to be changed, is to clear all trim/subtrim on transmitter, and connect to GUI. Note the value most channels idle at - this should be the number to choose. Once midrc is set, use subtrim on transmitter to make sure all channels (except throttle of course) are centered at midrc value.

To me it does not sound like mid_rc only is used for throttle...? My impression is that on futaba transmitters, the center point on all for channels, thr/yaw/pitch/roll, is 1520 and this setting helps to deal with that...?

theArchLadder, 2016-05-18:
The configurator normally shows 1500 for yaw/pitch/roll when the transmitter is not on. Changing mid_rc=1300 makes the configurator show 1300 on yaw/pitch/roll, so i think this setting is the expected center value from the transmitter?

Looks like we have to make a new setting if we want to have a percentage setting for neutral stick...

oleost, 2016-05-18:
Quote @digitalentity : "I think HOVER should be default - it's equal to MIDTHR if you didn't tune anything, and if you set your hover_throttle right you have smoother transitions to/from althold."

This assumes you already set your hover throttle right, which is hard to set an default value thats will fit different quads.

Thats why I would as default have:

nav_mc_hover_thr = 0 // Disabled (Use current throttle as starting point for hover)
nav_mc_netraul_althold = CURRENT

This one in my opinion be the best default, until you actually tune your quad with the right hover throttle etc. And if you already are tuning hover throttle you

digitalentity, 2016-05-18:
@oleost what happens when AltHold is enabled on the ground with your defaults?

oleost, 2016-05-18:
@digitalentity 

nav_mc_netraul_althold: As discuss earlier. There must be an minimum allowed stick input to allow for decend. So just directly set netraul at 30%

nav_mc_hover_thr: will simply start at zero throttle, and will wait for stick to go above netraul (30%) before starting to increase throttle.

So, the quad would stay still on ground, untill stick are moved beyond an hardcoded minimum value.

digitalentity, 2016-05-18:
@oleost and it will take forever until VEL_I grows from zero throttle to the point when quad can actually take off.

oleost, 2016-05-18:
@digitalentity https://github.com/iNavFlight/inav/commit/00630fe3ffee04396702d0c3ac5f5fb6d569c99d

Is there an possibilty to add so the codes detects "takeoff" mode an either jumps faster to 1200us/20% or have increased PIDs/gain until it detects takeoff?

digitalentity, 2016-05-18:
@oleost We should try to avoid any hardcoded values. I also try to make code as general as possible and avoid clumsy shortcuts with zero throttle etc.
How about forcing user to setup nav_mc_hover_thr?

Don't allow arming until certain variables are set?

oleost, 2016-05-18:
@digitalentity Forcing value might be one solution.

But I really like the idea of when already hovering, instead of jumping to an predefined throttle output just use the current, since most people are already hovering nicely when activating such feature.
And then it doesnt need any retuning when changing props, batteries, or loads like gimbal etc.

Anyone looked at how example ardupilot and librepilot are doing this?

digitalentity, 2016-05-18:
Don't forget that altitude controller is also used when doing RTH and it might be activated by failsafe or as an "oh shit" option. Throttle might be way off from hover and setting it to "current" might cause a crash.

theArchLadder, 2016-05-18:
So i just pretended to be a lazy new user that didn't read the wiki, and flashed inav and went for a fly

This is what the althold jump looks like on my quad
I bring it up to hover just above ground in manual mode and then activate althold
https://www.youtube.com/watch?v=Yfj2PEw-dMA
I would say it jumps 2-3 meter, or rather, it's a tiny jump and then ascends for a bit.
Then it slowly starts to descend because my throttle stick is only at 25%.

IMHO, nothing dangerous happens here, the jump is not violent nor extremely high, and the descent is so slow that there is plenty of time to raise the throttle. I doubt anyone has a much lower hover throttle than me.

Now to my point: I find it very unlikely that someone would say "hey, i'm happy with the fact that my quad jumps up 3 meters when i activate althold, but how do i fix the neutral throttle position?"
The jump is the main problem here, everyone would want to fix that, so why not fix stick position as a bonus?

@digitalentity 
> Don't allow arming until certain variables are set?
I would rather have the user ask "why does my quad jump/descend" than ask "why does my quad not arm?", the latter can have many answers. IMHO, we should only prevent arming if something dangerous can happend.

oleost, 2016-10-01:
Closing this one, current way is a good way to do it.

closed 2016-10-01 by oleost