iNavFlight/inav #223: Unable to make latest master

oleost, 2016-05-18:
Can't compile latest master with SPRACINGF3
Tried just making for Naze32 and it worked, still some errors.

My build enviroment is Windows 8, and setup like in the docs https://github.com/cleanflight/cleanflight/blob/master/docs/development/Building%20in%20Windows.md

http://pastebin.com/gze0xEHC <<<--- make TARGET=SPRACINGF3

http://pastebin.com/qWxKUEwc <<<--- make

digitalentity, 2016-05-18:
This is caused by #149.
@martinbudden, can you fix this please?

martinbudden, 2016-05-18:
@digitalentity , a fix for this was already on the way...

See PR https://github.com/iNavFlight/inav/pull/224 , this is the first part of the sonar initialisation cleanup. There are a couple of more sonar initialisation cleanup PRs on the way...

digitalentity, 2016-05-18:
Fixed by https://github.com/iNavFlight/inav/commit/4c3c8a1b4870f7d806cde07db668f1751b88ad40

closed 2016-05-18 by digitalentity