iNavFlight/inav #229: BUG: Issue with Aux/Mode CLI restore.

oleost, 2016-05-19:
When restoring from CLI dump the Aux / mode is not correctly restored.

What I need to do to get it restored correctly is copy and save cli dump, and then copy aux only and save.

This has been for awhile. atleast back to April 02.

Board = SPRACINGF3
How to try it yourself.

Try and copy and save this CLI dump http://pastebin.com/6awLdti6 (Airplane mixer)

Watch your missing some modes. In  my case I only had NAV POSHOLD, NAV RTH and PASSTHRU set aftert first restore.

Paste in only the Aux channels

`# aux
aux 0 0 0 1300 2100
aux 1 2 1 1700 2100
aux 2 25 0 1300 2100
aux 3 3 2 1300 1700
aux 4 10 2 1300 1700
aux 5 9 2 1600 2100
aux 6 11 1 900 1250`

And hit save, and then the rest of the modes are there aswell. (Arm, Horizon, Airmode)

I know I'm not the only one, here is another guy aswell: http://www.rcgroups.com/forums/showpost.php?p=34422766&postcount=4622

theArchLadder, 2016-05-19:
I have also had this problem, but for me it's even worse:
Some of the settings are not set when i paste in a full dump. I have not noticed any patterns yet, what settings are not set seems a bit random.

Is the configurator to fast so the FC can't keep up or something?

DzikuVx, 2016-05-19:
I also had similar issue. Also modes config. Also SPRF3

oleost, 2016-05-19:
@theArchLadder Think thats an different issue. Had it once and after a reflash that issue was gone. The symptons was that it was slow when I copied it in CLI. Normally its lighting fast when coping stuff into CLI.

theArchLadder, 2016-05-19:
@oleost I have had it multiple times.
My aux/flight modes has sometimes been only partially restored

sppnk, 2016-05-19:
I dont know if it is the same issue, but with cc3d and naze, sometimes after a reset (no after a flashing or restotring) I lost all the mode configurations. 

And other times everything go sloooow, I have to reset and then everything go fine.

stronnag, 2016-05-19:
This is a well known feature of all CF firmware variants, the firmware cannot process mode changes fast enough, the buffer overflows, data is lost.  This is why the mwptools cf-cfi (cli backup / restore tool) uses a baud rate and context aware inter-line delay. Slower than pasting into a terminal, but reliable.

Otherwise paste in small batches, particularly for the aux section.

theArchLadder, 2016-05-19:
@stronnag How about adding something similar to the iNav configurator?

stronnag, 2016-05-19:
You are the second person to ask! I really don't do javascript. But it is possible that something simple like a fixed 25ms delay after each new line would help.

theArchLadder, 2016-05-19:
sudo @stronnag do javascript :stuck_out_tongue_winking_eye: 
Have you tried if only using a lower baud rate helps?

stronnag, 2016-05-19:
not tried that. lowering the baud rate becomes a real mess, both to flash and after as it's back to 115200, so you have to remember to change it on the FC, then in the tool.

DzikuVx, 2016-05-20:
@stronnag @theArchLadder @oleost please give a try to iNavFlight/inav-configurator#7

oleost, 2016-05-20:
I just tested and confirm it worked.

DzikuVx, 2016-05-20:
OK, since this seems to be configurator issue after all, I'm closing this one. Please refer to iNavFlight/inav-configurator#6 for further discussion

closed 2016-05-20 by DzikuVx