iNavFlight/inav #230: ACC calibration won´t work if FC is mounted upsidedown or in another angle

gurkenfolie, 2016-05-19:
The ACC calibration doesn´t take the FC alignment in account, it is not possible to calibrate it as shown in the pictures when the FC is mounted in another angle or upside down.
It is only possible to calibrate the FC when the FC points to the directions as shown in the calibration pictures, makes calbration a little bit confusing.

digitalentity, 2016-05-24:
Yes, that's a known bug. I'll address it.

eziosoft, 2016-05-27:
Is it present also in 1.0.1? That would explain why PH doesn't work for me nicely.

Flugsicherung, 2016-07-06:
Today I testing V1.1 on my SPRacingF3 Deluxe. The Board is mounted 180° in YAW to connect the short motor cables to the FC. Calibration of Gyro doesn´t work in this configuration.

digitalentity, 2016-08-18:
@Flugsicherung can't reproduce.

digitalentity, 2016-08-18:
First acc calibration point should be FC in the upright position. If FC is mounted upside down this means first position should be UAV upside down (so FC would be upright). Minor bug, postponing for 1.3

Flugsicherung, 2016-08-19:
I don´t mount the FC upside down, I mount the FC 180° in YAW to connect short motor cables.

digitalentity, 2016-08-19:
Then there is no reason for calibration to fail, unless you don't follow the procedure - https://github.com/iNavFlight/inav/wiki/4.-Sensor-calibration#accelerometer-calibration-steps

oleost, 2016-08-19:
@flugsichdrung Are you doing it like this  https://youtu.be/HxWSn_noNgg?

Flugsicherung, 2016-08-19:
Thanks for the info. I did it like cleanflight and that was wrong :)

closed 2016-11-15 by digitalentity

oleost, 2016-11-15:
@digitalentity why did you close this?

digitalentity, 2016-11-15:
Because it's irrelevant. First position of calibration has to be FC upright and calibration works if this is done properly. 

We need a more user-friendly calibration, but that's a different issue.

oleost, 2016-11-15:
Fair enough, just wondered if you did it by mistake since you labeled it bug a while ago.

digitalentity, 2016-11-15:
I thought that I would rework the procedure only in formware, however now I think it should be done better from both firmware and UI sides. Not user-friendly calibration is not forgotten :smile:

DzikuVx, 2016-11-15:
I also have this topic somewhere in my mind in context of Configurator
support