iNavFlight/inav #234: "Nav Position hold" wont arm on SPF3 1.01 onwards

ujara, 2016-05-20:
Noticed my SP Racing F3 flight controller wont arm with Nav Poshold (position hold) after iNAV 1.01. Prior to that I see with my M8N GPS several satellites and signal strength of 7+ in cleanflight GUI. If I install say iNAV 1.1 RC3 for example I see many satellites with the same previous setup but signal strength is now 2 or less in cleanflight GUI  and now my flight controller board does not arm. Can some one please see what is the issue with iNAV post 1.01? Thank you.

theArchLadder, 2016-05-20:
Have you read the wiki? See "notes".
https://github.com/iNavFlight/inav/wiki#notes
All the signal strength bars are replaced by one HDOP. Your GPS seems to be working fine.

Not sure why you can't arm...
Have you calibrated your mag and accelerometer? We need a wiki page for "why won't it arm"...

DzikuVx, 2016-05-20:
@ujara 

* you can not arm with PosHold enabled. This is a precaution issue. You have to arm in Angle (or Horizon or Arcro) take off and then  enable PosHold
* To enable PosHold you need mag calibration
* Configurator reports HDOP instead of "signal strength". The lower HDOP the better, should be below 2 for nice GPS positioning

theArchLadder, 2016-05-20:
I use nav_extra_arming_safety = OFF because i like to live dangerously. I can arm in poshold etc, is this intended?

DzikuVx, 2016-05-20:
@theArchLadder according to code, `nav_extra_arming_safety=OFF` disables all INAV safeties about arming

ujara, 2016-05-20:
Got it but it appears the Signal strength is now counter intutive and yes I do have my mag and acclerometer calibrated . I guess the idea is to fly in Horizon or Acro mode and then switch to position hold along with say altitude hold. So after that if I wanted to do an RTH it would require a third position in the AUX switch to get the drone back to where it started.Thanks for the feedback.

closed 2016-05-20 by ujara