iNavFlight/inav #242: RTH: Replace pre-landing-delay with wait-for-correct-heading

theArchLadder, 2016-05-24:
Right now there is a 5 second delay from when the copter reaches home position to the descent starts. Some delay is needed so the copter has time to yaw and get the right heading before starting to descend.

I use a mag_hold_rate_limit of 180dps, i want rapid heading change, it just looks and feels right on my racing quad.
On the other hand, @DzikuVx was happy with using 25dps on a big and fancy quad to make yaw smooth and gentle.

For my quad turning around should take just 1 second, so a 5 second delay feels annoyingly long for me, but is to short for @DzikuVx quad that takes over 7 seconds, this makes it tricky to set a good default pre-landing-delay value, and having to tune yet another setting sucks.

So my idea is to just waiting until the heading error is less than 5 degrees or something like that before starting to descend.
This would make it work with both high and low mag_hold_rate_limit values without any tuning required. And if the quad comes in with the correct heading we don't need as long delay. We could have a minimum delay of like 1 second if needed.

Thoughts?

digitalentity, 2016-05-24:
Good idea. I'll add this change and also ensure we have a minimum delay to settle down before starting descent.

DzikuVx, 2016-05-26:
As a matter of fact, I have nothing against descend while still changing heading.

theArchLadder, 2016-05-26:
@DzikuVx 
I have tried both, changing heading while descending saves some time, but i think it sort of looks nice to settle in and change heading before descending... I can't make up my mind. Maybe we should have an option for this?

DzikuVx, 2016-05-26:
@theArchLadder this can be easily made configurable with CLI variables

PR #320 (2016-06-24): RTH: Replace fixed pre-landing delay with waiting for correct heading
Fixes #242
closed 2016-06-24 by digitalentity