iNavFlight/inav #245: RTH: Wrong descent velocity when activated within nav_min_rth_distance

theArchLadder, 2016-05-24:
Credit goes to @stronnag for discovering this bug.
When descending from high altitude, the vertical target velocity should start at -200 and ramp down to -50, but with this bug the target velocity is at -50 all the time when RTH was activated within nav_min_rth_distance

I did a test flight with some navDebug[] here:
https://github.com/iNavFlight/inav/blob/master/src/main/flight/navigation_rewrite.c#L957
posControl.homePosition.pos.V.Z should have been -47, but was 750, where did this number come from?

posControl.homePosition.pos.V.Z was correct (equal to arming altitude) when doing a RTH from outside nav_min_rth_distance.

theArchLadder, 2016-05-24:
Oh, looks like i activaed RTH at exactly navPos[2] = 750
So i guess for some reason the current altitude is taken as the home altitude...

theArchLadder, 2016-05-24:
Home position gets reset here:
https://github.com/iNavFlight/inav/blob/master/src/main/flight/navigation_rewrite.c#L827

Do we really want to reset it? How about just skip the altitude-change and go straight to home position, and then descend?

stronnag, 2016-05-24:
Start at https://github.com/iNavFlight/inav/blob/master/src/main/flight/navigation_rewrite.c#L744
`// If close to home - reset home position`
end up at https://github.com/iNavFlight/inav/blob/master/src/main/flight/navigation_rewrite.c#L1493
where we blat the altitude, because we accept current pos as home (including altitude). At least we don't instantly disarm!

theArchLadder, 2016-05-24:
Agreed, the home position for altitude should not be reset! This would solve the bug.

Let me clarify what i tried to ask before:
Copter is 4 meters sideways away frome home, RTH is activated, what should happend?
1. Just land (as it is now) 4 meters away from home position.
2. Keep altitude and go to home position, then land at home position.

I think i would vote for 2, IMHO RTH should always try to land at home position...

stronnag, 2016-05-24:
I don't mind very much which happens, as I'm invariably over home when I do this. I'd just like that (a) it does not ascend as there should not be vertical obstructions and (b) it does not do the whole descent at 'slow'.
I guess (2) is preferable.

digitalentity, 2016-05-25:
The whole point of min RTH distance is to avoid mishaps when RTH is
triggered on the ground (after arming but prior to flying the machine) for
whatever reason.

Initiating RTH regardless of distance may cause the machine to tip over or
jump in altitude so I'd rather keep the "land where you are" behavior.

However the slow decent bug must be fixed and not resetting the home
altitude seems the right way to solve the issue.

digitalentity, 2016-05-25:
Actually same altitude bug may affect the "home reset" functionality.

theArchLadder, 2016-05-25:
@digitalentity That's a good argument, you have convinced me :+1:

closed 2016-09-01 by digitalentity