iNavFlight/inav #248: Add support for all-in-one board (STM32F303 + MPU9250)

csarka, 2016-05-26:
It would be really great to have support for this very convenient all-in-one board.  We (the community) already have all the parts of coding, it's just a matter of putting them together.  Upstream Cleanflight has the necessary drivers for MPU-6500 and AK8963 contained within MPU-9250.

deviantspeed from RC Groups has been able to adapt SPRACINGF3 from Betaflight 2.7.0 as noted here: http://www.rcgroups.com/forums/showthread.php?t=2633525&page=11

I have attached a diff of the changes made by deviantspeed.  I have tried to mimic his changes in a test version of inav 1.1, however I have been unsuccessful so far.
[deviantspeed_changes.txt](https://github.com/iNavFlight/inav/files/284911/deviantspeed_changes.txt)

The board is sold by Banggod (Deluxe version) and RTFQ:
http://www.banggood.com/SP-Racing-F3-Flight-Controller-With-Integrated-OSD-for-FPV-Multicopter-p-1032369.html
http://www.readytoflyquads.com/flip-32-f3-osd-791

CPU: STM32F303 with 256K ROM
MPU-9250 
Flash Memory: 16M for Black Box
BMP280 barometer

The MPU-9250 contains a MPU-6500  for accelerometer and gyroscope and an AK8963 for the compass (datasheet https://store.invensense.com/datasheets/invensense/MPU9250REV1.0.pdf )

Linjieqiang, 2016-05-27:
why not use sd-card modle?

csarka, 2016-05-31:
Also, samuelpowell has been able to adapt betaflight here:
https://github.com/samuelpowell/betaflight/tree/F32F3OSD
However, he says has not been able to get the MAG working.

csarka, 2016-08-04:
The MPU-6500 gyro is too noisy in this board and causes a lot of vibration problems and twitching in the quadcopter.  I'm recommending we forget about this idea.

closed 2016-08-04 by csarka

Linjieqiang, 2016-08-04:
But you can try ICM gyro?

Linjieqiang, 2016-08-04:
What about this chip?https://www.invensense.com/products/motion-tracking/6-axis/icm-20602/

csarka, 2016-08-04:
@Linjieqiang thank you for your suggestions.  However, I think an addition of an external gyro to this board defeats the purpose of an "All-in-one" board.

Linjieqiang, 2016-08-04:
So cool.But what sensors do you want to use>