iNavFlight/inav #251: CLEANUP: LEDSTRIP code cleanup

digitalentity, 2016-05-28:
Unify DMA handling code, catch-up with CF

DzikuVx, 2016-05-28:
I was looking what CF done with LED and it will be big task. There are many, many changes and commits there

digitalentity, 2016-05-29:
Yes, but we might be able to save some space and also clean up the hardware
part of the code.

martinbudden, 2016-05-29:
I think it is worthwhile to include this. I reckon it will save around 500 bytes of ROM, which might just be enough to include the LED code in the CC3D build.

shaw05, 2016-06-08:
Would also supercede issue #199

majik12345, 2016-06-11:
The LED changes dont appear to be too much work, but in any case I would suggest before any more new features are added or worked on catching up with the current cleanflight commits should be a priority

gaelj, 2016-06-22:
Parameter groups would be nice to have first, for one thing.

digitalentity, 2016-06-22:
@gaelj it's peing slooooooowly worked on in #151

skaman82, 2016-07-09:
@digitalentity Can you include the current LED functionality from CF in 1.2 or is it too much work? Having the GPS state in LED Strip is realy a nice thing. Blink Mode too.

gaelj, 2016-07-12:
It can be easily adapted from the Betaflight version, which does not use PGs.

closed 2016-07-29 by gaelj