iNavFlight/inav #252: Set nav_fw_cruise_thr below nav_mc_hover_thr cause RTH drop/crash

kronosol, 2016-05-28:
Hey, i´m quiet new here.
Found an issue that cause **min throttle** drop of the quad, if RTH turned on.
For example my settings was:
*nav_mc_hover_thr = 1500
nav_fw_cruise_thr = 1400*
After i changed the settings to,
*nav_mc_hover_thr = 1430
nav_fw_cruise_thr = 1500*
the RTH mode worked well.
I´m quiet safe with flying quads, but maybe a beginner can´t turn off RTH fast enough and rescue the quad. I would prefer a safety function that cruise thr. can´t below hover thr.
I´ll try do record a blackbox log of that, but maybe someone can repeat that?
1. RTH turned on > quad point home pos.
2. Quad throttle down till minthr. in within 3 sec.
3. RTH turned off > quad can be rescued with your acro skills :+1:

digitalentity, 2016-05-28:
This is quite weird. nav_fw_cruise_thr setting is not used for
multicopters, it's airplane-specific. Blackbox log of the drop would help
tracking the issue down.

digitalentity, 2016-05-28:
Also a full config dump may give us a hint about what's going on.

kronosol, 2016-05-28:
Ok, what kind of blackbox logs do we need?
I´ll set my settings back and do a test flight tomorrow in the evening.

digitalentity, 2016-05-28:
Blackbox log with record of the issue.

kronosol, 2016-05-29:
Ok, (*nav_mc_hover_thr = 1500 & nav_fw_cruise_thr = 1400*) **wasn´t the Problem!**
I found the issue/bug? comes out if *(nav_rth_alt_mode = FIXED & nav_rth_alt = 1000)*,
**but**
1.Only if the quad stays **not** about the Home Position, (altitude doesn´t matter).
2. The quad is higher than *nav_rth_alt = 1000*
So I created a dropbox folder with a Readme, Video, Blackbox Log and CLI Dump.
[DropboxRTHIssue](https://www.dropbox.com/sh/fetv6ofdxhz5myq/AAAaLONEJyNtnsRoJiE6HjvHa?dl=0)
PS: Did some major changes to the Quad and didn´t pid tune. So don´t care about that wobble and bubble of Gimbal and Quad, i will fix that now i know what caused the drop.

theArchLadder, 2016-05-30:
The problem is that there is no limit on target descend velocity in this case, navTgtVel[2] goes to -1200cm/s (very fast, even -200cm/s is fast). You aborted 15m over the 10m target height, so navTgtVel[2] never slowed down. It might have worked fine and just slowed down when it got near 10m of altitude, or it would have had to high velocity and crashed...

kronosol, 2016-05-30:
I´m sure it had to high velocity. 
The Quad was falling from around 60m to 15m in around 3 seconds(54km/h).
But in my opinion the real problem is, this can happen on every quad.
A beginner might set *rth_alt_mode = Fixed* on standard settings and can´t catch out the quad form crashing.
Maybe someone can reproduce this RTHIssue so we are clear it´s a code problem not my fault?

digitalentity, 2016-05-30:
Ok, so we have an issue here that there is no limit for climb/descent rate
in auto modes. This needs fixing. Thanks for finding this out!

theArchLadder, 2016-05-30:
@kronosol To fast velocity for what?
Is the velocity faster than what looks and feels safe? Yes!
Did vertical velocity go faster than target velocity? Nope.
Is it to fast for the copter to be capable of slowing down and not go under target altitude? No, i don't think it is, braking distance with 1 G from 54km/h (15m/s) is only 11,48 meters, you where more than 15 meters above target altitude, but still, this is way to agressive and should be fixed. And i doubt the current vertical velocity pid-controller would handle a situation like this without trouble.

Nothing really went wrong in the code and the copter would in theory be capable of this (not sure if inav is though, don't try it!). I'm just saying this issue is more about making sure we don't have a heart attack rather than fixing something broken in the code. :stuck_out_tongue_winking_eye:

kronosol, 2016-05-30:
@theArchLadder Nice argument, but i see it like this.
I turned RTH on, Quad is falling as seen in the video, i disabled RTH maybe 20m above the Ground. But 1. the quad was accelerating all the time, 15m/s wasn´t the end.
2. The Inav code will "**maybe**" start to slow down at 10m, but because of ***fancy latency*** we only 
**measure 10m**, while lets say the quad is **8 m high**. 8m/15m/s = 0,53 sec for the code to stop the quad before it crashes into the Ground.
Maybe this will work with a *nav_rth_altitude = 2000*, but i´m not the one who test it :+1:

theArchLadder, 2016-05-30:
@kronosol I just tried it on my 250 racer just for fun (it's crashable!).
I used nav_rth_alt = 500. I went up to 15-20m and the activated RTH, it fell down and stopped i would say 2 meters under the target altitude, so it didn't crash but it did go under target altitude... This was on a powerfull racing quad with higher than default tuning for altitude controller though.

Not sure if a -200cm/s hard limit is the right answer either (like it is on rth landing stage), from 500m it would take over 4 minutes to land (my battery would be toast and my quad would fall out of the sky)... Some youtuber went up to 5000m with a quad, then it would take 42 minutes to come down!

EZ-GUI said it saved the blackbox log but it looks like it didn't, crap...

kronosol, 2016-05-31:
@theArchLadder Yes your´re right!
Once i disarmed my racing quad (noFPV/XFrame) with stick command in mid air.
But because of the heavy battery it stabilized on its own and did something like "auto-rotation".
This was really interesting, because there was a point with no more acceleration. 
In the end, not even a Prop dies. (Falling from 100m), but the frame go a little bit curved.
(I couldn´t re-arm because at these point i did´t know the reason for crashing)
This could be a never seen feature for coming down on very high altitudes (>100m), but maybe the ESC would be burned because of the generated electricity and it only works on quads with a low center of gravity. But if you are higher than 500m,.. maybe you can use this to come down without using power, than on 100m you re-arm and do these cool RTH things.. :yum:

kronosol, 2016-05-31:
It might be a nice thought,.. 
If your Quad is falling on "Autorotation" and all props would be freewheeling, the way to stabilize a quad with a low center of gravity would be slowing down the props.
Like the opposed way we´re doing it now on normal flight,.. but we get off the topic here...

digitalentity, 2016-07-23:
Fixed by #257

closed 2016-07-23 by digitalentity