iNavFlight/inav #253: FlySky telemetry

sinokop-maksim, 2016-05-29:
please examine the possibility of telemetry output to the FlySky.
P.S. not fRsky.
Thank you

digitalentity, 2016-05-29:
Good idea, but who's going to develop the code?

Currently I have too many things to attend to, so this is going to be
postponed unless somebody will do the coding.

DzikuVx, 2016-05-29:
i-bus telemetry is also waiting to be merged to CF in cleanflight#1723 
I was thinking about porting it, but recently sold all mine FlySky hardware

digitalentity, 2016-05-29:
I can test the code for you when I'm back home in 3 weeks. I have a FlySky
TX and RX at home.

sinokop-maksim, 2016-05-31:
I can currently test, give the hex file and tell me where to connect wires.
p.s. I have sprasing f3

DzikuVx, 2016-05-31:
@digitalentity @sinokop-maksim OK, I will try porting it. Let's see what will happen :)

DzikuVx, 2016-05-31:
@sinokop-maksim FlySky iBus telemetry enabled build for SPRacingF3 can be dowloaded [here](https://drive.google.com/file/d/0Bx6X245zlrsnTW14ZTVYemtqcWM/view?usp=sharing)

Absolutely not tested, no idea if it works or machine is flyable. So, please test :)
Just like in original CF version, iBus telemetry has ID 256, so
`serial 1 256 115200 57600 115200 115200`
should enable it on UART2

sinokop-maksim, 2016-05-31:
unfortunately it did not work :(

DzikuVx, 2016-06-01:
Then WE would have to wait for @digitalentity

sinokop-maksim, 2016-06-01:
while we wait for @digitalentity, please hex file the flip32.
possible to flip32 will work.
and I know that I have not messed with the wires on sprasingf3 :)
Thank you.

digitalentity, 2016-06-01:
@sinokop-maksim @DzikuVx 
I'll have a look at the code today. Unfortunately it's another 3 weeks until I get my hands back on my FlySky TX to test anything on real hardware.

sarpba, 2016-06-01:
(first of all sorry for my bad english)

It's absolutly working in my config.
SP raceing F3 (clone)
TGY-I10 TX
FS-IA6B RX

One problem only, but it's not realy big problem. Need to config it manual, from CLI, because the IBUS Telemetry menu don't appelar in the configurator ports menu.

In my config the IBUS RX is on the softserial1 prort becouse i destroyed the UART3 RX port. SO I use UART3 TX port for telemetry.

CLI commands:
serial 2 256 115200 38400 115200 115200
save

After turn on the telemetry in Configuration tab.

BTW it's amazing! THANK YOU!

DzikuVx, 2016-06-01:
OK, so we have 1 case working, 1 case not working. 
@sinokop-maksim maybe it was wiring problem? Either way, I will prepare hex for Flip32 with ibus telemetry enabled in course of today

DzikuVx, 2016-06-01:
@sinokop-maksim Flip32/Naze32 with only IBUS telemetry is available [here](https://drive.google.com/file/d/0Bx6X245zlrsnaTR3VWtLXzlhSjg/view?usp=sharing)

sarpba, 2016-06-01:
One big problem.

The TX randomly drop the connection. :/ I updated the TX/RX yesterday to 1.0.8.0. I hope the new firmware is the problem. I'll try a downgrade.

sinokop-maksim, 2016-06-01:
I confirm the work on the spracinf3.
My mistake was that I do not connect the + and - only the signal wire :(

sinokop-maksim, 2016-06-01:
and the like can be moved in line sensors?
eg I need a first voltage of a second gps the satellites, is this possible? or configuration data only in aparatury? and the temperature I do not need :)

DzikuVx, 2016-06-01:
Problem with iBus telemetry is that this is polling protocol. Receiver decided which data it wants and in which order. By design it is to poll FlySky sensors. So, if FlySky does not make a sensor of a type, radio will not poll for such a info.

sinokop-maksim, 2016-06-01:
I understand you correctly, you are not able to create a virtual sensor?
and how to substitute data for the sensor?
I need the data telemetry  gps satellites, distance, voltage on the battery.
temperature, motor speed I do not need :) to replace them can get the data on the satellites and gps distance?
p.s. Sorry for my english. a Russian translation Google :)

DzikuVx, 2016-06-01:
It is possible to replace data, why not. But rather not as permanent solution.

To creat virtual sensors I have no idea. No FlySky devices on my desk

sinokop-maksim, 2016-06-01:
if I upload Screenshots virtual sensors.
so you can do? I understand that the longer and harder to develop, but it's better than nothing.

DzikuVx, 2016-06-01:
I can not develop this, I do not have FlySky radio

sinokop-maksim, 2016-06-01:
very sorry that you live in Poland, if only you lived in Ukraine, I would have sent you my flysky for tests (with return later) :)
do you plan to buy their own fliski? it makes sense to write to you for the development of this issue?

sinokop-maksim, 2016-06-01:
possible to replace the sensor data via the CLI ?
if you can not add new sensors, they might replace data.

DzikuVx, 2016-06-01:
No idea, and I do not want to promise anything that I might not keep

sarpba, 2016-06-01:
Disconnection problems than anyone else?

Turnigy TGY-I10 TX - FS-IA6B

The downgrade to 1.0.7.0 is not helped. Always fully disconect in 0-5 minutes.

FC is SPraceing F3 or Naze32... Same.

DzikuVx, 2016-09-15:
ibus telemetry has been ported and merged ( #259 ) , let's close it

closed 2016-09-15 by DzikuVx