iNavFlight/inav #255: Althold: Allow using GPS for vertical velocity on multicopters

theArchLadder, 2016-05-29:
My althold is nervous with inav_w_z_baro_p=0.35, I'm happy with the althold at inav_w_z_baro_p=0.2, but if i activate althold lets say 2 meters from the ground and fly around, it will almost fall into the ground and drift way higher when stopping etc.

@iforce2d You also had this problem, right?

Using inav_w_z_gps_v=0.5 or so makes this problem almost go away completely, and i have not noticed any bad behavior so far, no drift etc. This is what it looks like during poshold:
https://www.youtube.com/watch?v=U4hAxSl03-U

I also tried using gps for vertical position, inav_w_z_gps_p, but that drifts several meters up and down very slowly.

I removed `STATE(FIXED_WING)` from here to be able to use gps:
https://github.com/iNavFlight/inav/blob/master/src/main/flight/navigation_rewrite_pos_estimator.c#L519
But we would need a separate setting from airplanes for this so the default could still be zero.

iforce2d, 2016-05-29:
I think the problem I mentioned was dropping altitude when making horizontal movement inputs during poshold. I was mainly noting that the altitude is held much better when moving during autonomous modes.

I'm not sure how your video demonstrates your problem - what was it like before? I've never had any trouble with static position hold in very calm conditions like that (SBAS problems aside of course :)

theArchLadder, 2016-05-29:
I also have a bit of dropping altitude with horizontal velocity, this problem seems to mostly go away when using gps though.

My mistake posting that video, as you said it doesn't demonstrate the issue, @digitalentity has said that gps drifts to much to be usable for altitude, i just wanted to show that i do not have that issue...

theArchLadder, 2016-05-29:
@iforce2d Perhaps you would like to try this and see if it's helpful for your problem? Would be interesting to know.

All you need to do is this:

Replace:
`bool useGpsZ = STATE(FIXED_WING) && isGPSValid;` 
With:
`bool useGpsZ = isGPSValid;`
Here: https://github.com/iNavFlight/inav/blob/master/src/main/flight/navigation_rewrite_pos_estimator.c#L519

Change these settings:
set inav_w_z_gps_p = 0
set inav_w_z_gps_v = 0.5, to start with, but play around with it, up to 2 or so i guess.

kronosol, 2016-06-20:
I know this sounds basic, but did you covered your baro from sunlight? In my case I've never had problems with this till the last days,.. I did some changes and used white foam to protect the baro. But while turning the quad, on one point the sunlight touched the foam and the quad drifted around 10m of altitude! First I doesn't mentioned the sunlight because there where some clouds on the sky, buy I covered with some tape and now it's rock solid again!

digitalentity, 2016-07-12:
@iforce2d @theArchLadder 
Using GPS for climb rate estimation is always allowed in the above PR

PR #346 (2016-07-24): Use GPS for vertical velocity estimation on multirotors
Closes #255
closed 2016-07-24 by digitalentity