iNavFlight/inav #260: Gyro_sync causes motors to spin on power on

DzikuVx, 2016-06-03:
I've seen something scary today. With gyro_sync=ON motors started to spin when power connected. Good thing they were spinning on low power... Configuration

* SPRacingF3
* OneShot125
* GPS
* OLED display

Disabling gyro_sync fixed the issue

theArchLadder, 2016-06-04:
Strange and scary! I'm using the same setup except the oled, and that has never happened for me. You use aux arming, right? I use stick arming...

Did it happen on every startup or just once?

DzikuVx, 2016-06-04:
On every startup. And yes, aux arming.

DzikuVx, 2016-06-04:
And BTW it was gyro_sync_denom=1

theArchLadder, 2016-06-04:
I also use
```
gyro_sync=ON
gyro_sync_denom=1
```
Perhaps you could try and see if you still have this problem with stick arming, or without oled?
I have used aux arming on some occasions, but never had the motors spin up when i connected the battery...

digitalentity, 2016-06-04:
Weird. Can you please test if it's related to oled or some other feature?
I'm flying oneshot with gyro sync at 1khz - no issues.

DzikuVx, 2016-06-04:
I will try to confirm the cause.

DzikuVx, 2016-06-08:
So far, as I do not have access to any F3 board, I was unable to confirm that on Flip32 with display, GPS, gyro_sync=ON and denom=2 since firmware prohibits looptime faster than 2000us

DzikuVx, 2016-06-27:
Although I'm on vacation, I took my 250 and since it's raining I've decided to update few things and unfortunately I had the same issue one more time. This time on completely different setup:

* SPRacingF3 (Chinese)
* Emax Nano 12A ESC (They say it's BLHeli by EMAX)
* OneShot125
* No GPS, Display

After setting looptime 1000
```
set gyro_sync = ON
set gyro_sync_denom = 8
set gyro_lpf = 256HZ
```

Motors started to spin on boot. Setting `set gyro_sync_denom = 12` (looptime 1500) problem went away. 

In both cases it was EMAX ESC with stock firmware. So I tend to think it is something with ESCs by EMAX.

DzikuVx, 2016-09-24:
I'm closing this since I've somehow confirmed this issue is EMAX ESC only

closed 2016-09-24 by DzikuVx