iNavFlight/inav #267: CC3D & PWM receiver compatibility issue

digitalentity, 2016-06-12:
It appears that CC3D is incompatible with PWM receivers. When PWM RX is used channels are mixed up and output 5 gets disabled.
I suspect this code: https://github.com/iNavFlight/inav/blob/master/src/main/drivers/pwm_mapping.c#L251
This code originates from 30c550b83dfa7a001165916043dc8cccf9f1a276. I'm not sure the timer clash is a problem when OneShot and Brushed motors is not used. What do you think?

BTW: CC3D+PWM RX is cheapest possible solution for airplanes :smile:

martinbudden, 2016-06-13:
It should be possible to fix this.

BTW: CC3D+PWM RX may currently be the cheapest possible solution for airplanes, but I'm working on a cheaper solution CC3D + NRF24L01 RX, see: https://github.com/martinbudden/cleanflight/tree/inav_nrf24

This is almost ready for a PR.

digitalentity, 2016-06-13:
@martinbudden 
I have a proof of concept fix verified to work (CC3D+PWM+Airplane). Soon I'll shape it into a PR and merge.

NRF24L01 is indeed the cheapest possible solution. Do we have compatible TX-es?

martinbudden, 2016-06-13:
As for TX-es there are a number of possibilities.

1. Use http://www.deviationtx.com software on a modified Walkera Devo transmitter (this is what I use)
2. I've implemented some of the protocols used by the transmitters for toy quadcopters (e.g. Syma), although these transmitters have limited ranged. The toy transmitters are available for £10 to £15 though. There might be a toy transmitter with good range available - I haven't researched this yet.
3. Build your own. Or rather modify a cheap toy transmitter by adding an arduino and NRF24. Not as difficult as it sounds - @iforce2d has done this and has a video on youtube.

digitalentity, 2016-06-13:
I should investigate this. If range can be boosted to at least 1 km this will make it a perfect solution for a poor man's airplane :smiley: 

I see you have software SPI in your branch. This will come in very handy also for optical rangefinder (SPI as well).

BTW: Is it possible to get RSSI out of NRF24L01? If so it will be also a very good mid-range solution (with directional antenna on the TX)

martinbudden, 2016-06-13:
See https://www.youtube.com/watch?v=gtM832Z0ujE for @iforce2d 's nRF24L01 range comparison. 1km is indeed possible.

I'm not sure if RSSI is available. I don't recall seeing it when I read the spec document, but I'll have another look.

digitalentity, 2016-06-13:
No, according to the datasheet nRF24L01 doesn't have RSSI, but we can create a "packet loss" indication as a substitute for RSSI.

Ordered myself a few nRF24L01 modules to test. CC3D + nRF RX + BN880 GPS is going to be a very cheap GPS-capable solution for an airplane! And relatively long range as well.